MicrochipTech/cryptoauthlib #2: linux_i2c_userspace.c "hal_i2c_wake" returns ATCA_RX_NO_RESPONSE

cdbierl, 2018-02-06:
Greetings,

I am using the linux_i2c_userspace HAL with the 508A chip. I've update the 508A configuration struct to match my slave address of 0x60 and bus of 1. When running the cmd-processor, I submit the "508" command to select the desired config. When I then execute the "sernum" command I am presented with an error: "atcab_read_serial_number() failed with ret=0x000000E7". I figured I'd start with something basic before even attempting the unit tests.

I traced this to the HAL's wakeup function. The read function is returning -1 and thus returning a status of ATCA_RX_NO_RESPONSE. i2cdetect does find my device at 0x60. Our device tree file has the i2c bus setup for the standard 100kHz. 

Thank you for any guidance you can provide.

bryan-hunt, 2018-02-06:
I2C addresses are 7 bits. The 508 uses the upper most 7 bits of the address byte to identify that address while Linux uses the lower 7 bits. The address 0x60 as identified by linux when shifted left is 0xC0 which is the default address of the 508.

Try change the configuration structure back to the default 0xC0 and let me know the results.

bryan-hunt, 2018-02-06:
As an FYI I don't recommend running the various tests as they provide little value and you'll get numerous failures if your device has not been configured (which is a one time operation). 

The commands info, sernum, readcfg, & lockstat are sufficient to verify the library is communicating with the device.

cdbierl, 2018-02-07:
Thank you for the quick reply. Rookie move on my part! Changing the slave address back to the original 0xC0 worked and sernum returns "01 23 AA 75 B3 17 A9 EE EE". Thanks again!

bryan-hunt, 2018-02-07:
Glad to see that did the trick. Let us know if you encounter any other issues.

closed 2018-02-07 by bryan-hunt