MicrochipTech/cryptoauthlib #3: node-auth-basic sample: client-provision command fails for first time

cdbierl, 2018-02-09:
I have managed to build the node-auth-basic sample, released on 2017-11-17, against the latest version of cryptoauthlib. I have been able to run nearly most of all the unit tests successfully.

I am trying to run through this guide: http://ww1.microchip.com/downloads/en/AppNotes/Atmel-8983-CryptoAuth-ATECC508A-Node-Example-Asymmetric-PKI-ApplicationNote.pdf

While attempting to perform the client-provision command for the first time, I received this:
Warning: PrivWrite to slot 7 failed. Example may still work though.
client_provision failed with error code 44

Is there anything I might need to undo from the unit tests?

Thank you for any assistance,
Chris

BenUdall-Microchip, 2018-02-09:
Hi Chris,
The short answer is an ATECC508A device used for the unit tests can not be used with the node-auth-basic example and vice-versa. You'll need to use a new ATECC508A device for each example.

The unit tests, node-auth-basic, and other example projects often require different configurations for the ATECC508A device. However, once configured, it can not be reset and reconfigured, due to security concerns.

This limitation does make it challenging to experiment with. We often recommend one of the AT88CK101 socket boards (https://www.microchipdirect.com/product/search/all/ck101) and a handful of blank parts when going through the initial experimentation and development phases.

Regards,
-Ben

cdbierl, 2018-02-09:
Understood. I have a AT88CK590 at my disposal already, which I probably should have jumped to after verifying i2c communications.  Will the AT88CK590 allow for experimenting with unit tests and the node-auth-basic sample? In other words will it be able to be returned to a clean slate?

BenUdall-Microchip, 2018-02-09:
Hi Chris,
Unfortunately not. The AT88CK590 just has a ATECC508A, ATSHA204A, and ATAES132A available, but the limitation resides within the ATECC508A device itself. Once configured, it can't be reset.

closed 2018-02-16 by BenUdall-Microchip