MicrochipTech/cryptoauthlib #5: ESP32 IDF support

oemdaro, 2018-03-29:
Issues Type: Question

Can we use it with ESP32, ESP-IDF (Espressif IoT Development Framework)?

bryan-hunt, 2018-03-29:
We don't currently have an I2C driver written for the ESP32. We are aware of the desirability of an ESP32 integration. Do you have a Microchip FAE contact already? If you do you can send them a request but otherwise you can make one through the http://www.microchip.com/support portal.

We do try to keep cryptoauthlib simple and ansi-c so for the moment hopefully the porting exercises for you should be straight-forward and only require you to develop the I2C driver.

copercini, 2018-06-14:
I'm also looking for it

sukeshak, 2018-07-23:
I am also looking for it :)

nagarathna321, 2018-08-31:
can any one help me to create I2c driver for esp32 please 
replay me as much as possible

bryan-hunt, 2018-10-26:
An ESP32 Hal driver has been added to the library now which you can see in:
https://github.com/MicrochipTech/cryptoauthlib/blob/master/lib/hal/hal_esp32_i2c.c
https://github.com/MicrochipTech/cryptoauthlib/blob/master/lib/hal/hal_esp32_timer.c

In order to use it does require a bit of set up that I've tried to detail below. If you're familiar with how esp-idf project build this should be fairly familiar:

Copy cryptoauthlib into your esp32 project and in the cryptoauthlib folder add a component.mk file:

```
CRYPTOAUTHLIB_DIR := lib

COMPONENT_SRCDIRS := $(CRYPTOAUTHLIB_DIR)/atcacert \
                     $(CRYPTOAUTHLIB_DIR)/basic \
                     $(CRYPTOAUTHLIB_DIR)/crypto \
                     $(CRYPTOAUTHLIB_DIR)/crypto/hashes \
                     $(CRYPTOAUTHLIB_DIR)/host \
                     $(CRYPTOAUTHLIB_DIR)/mbedtls \
                     $(CRYPTOAUTHLIB_DIR)

COMPONENT_ADD_INCLUDEDIRS := $(CRYPTOAUTHLIB_DIR)

CFLAGS+=-DESP32 -DATCA_HAL_I2C -DATCA_PRINTF
```

Create a seperate dummy folder named cryptoauthlib_hal and create another component.mk file:

```
CRYPTOAUTHLIB_DIR := ../cryptoauthlib/lib

COMPONENT_SRCDIRS := $(CRYPTOAUTHLIB_DIR)/hal

COMPONENT_OBJS := $(CRYPTOAUTHLIB_DIR)/hal/atca_hal.o $(CRYPTOAUTHLIB_DIR)/hal/hal_freertos.o $(CRYPTOAUTHLIB_DIR)/hal/hal_esp32_i2c.o $(CRYPTOAUTHLIB_DIR)/hal/hal_esp32_timer.o 

COMPONENT_ADD_INCLUDEDIRS := $(CRYPTOAUTHLIB_DIR)

CFLAGS+=-DESP32 -DATCA_HAL_I2C -DATCA_PRINTF -DATCA_USE_RTOS_TIMER

cryptoauthlib/lib/hal/hal_freertos.o: CFLAGS+= -I$(IDF_PATH)/components/freertos/include/freertos
```

In the root project directory add the following to your component.mk file (or create it if it does not exist)
```
EXTRA_COMPONENT_DIRS := cryptoauthlib cryptoauthlib_hal
```

Generally your file structure would look like the following when you're done setting it up:

```
<project root>
    cryptoauthlib/
        lib/
        component.mk
    cryptoauthlib_hal/
        component.mk
    app_main.c
    component.mk
```

oemdaro, 2018-11-06:
Really really thank @bryan-hunt .
Thank for your hard work. I will try it asap.

closed 2019-01-09 by bryan-hunt

PBearson, 2019-01-31:
@bryan-hunt Can you provide an example of an application that uses the cryptoauthlib in a really simple manner? (Maybe generating a random number or reading the crypto chip's serial number...)

The following code crashes when I try to run it on the ESP32 (it compiles fine):
```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"
#include "cryptoauthlib.h"

void app_main()
{
        ATCAIfaceCfg cfg = {
                .iface_type             = ATCA_I2C_IFACE,
                .devtype                = ATECC608A,
                .atcai2c.slave_address  = 0XC0,
                .atcai2c.bus            = 2,
                .atcai2c.baud           = 400000,
                .wake_delay             = 1500,
                .rx_retries             = 20
        };

        uint8_t rand_out;
        atcab_init(&cfg);
        atcab_random(&rand_out);
}
```

This is the error I get:
[image]

bryan-hunt, 2019-01-31:
@PBearson See the documentation for [atcab_random](https://microchiptech.github.io/cryptoauthlib/html/a00916.html#ga4fa442396dfbb89e59dfca837b565cec) which returns 32 bytes. You're overwriting the stack (by quite a lot) in this example.

bryan-hunt, 2019-01-31:
@PBearson What is the return code from atcab_init? Is the stack size sufficient? Is I2C enabled?

PBearson, 2019-01-31:
@bryan-hunt adcab_init returns error code 225, which I think is ATCA_GEN_FAIL. It could be that I2C is not enabled. I'll look into it.

PBearson, 2019-02-04:
@bryan-hunt Just letting you know that the I2C connection was indeed the problem. Following [this example](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/i2c/i2c_self_test/main/i2c_example_main.c), I was able to initialize I2C first before using the crypto library. The atcab_random() command now successfully returns a random value.

Thanks for your help!

JavierPascual, 2019-05-10:
Hi! I'm using ATECC608A with ESP32 (i2c) but I'cant connect. Always return error 240 (atcab_read_bytes_zone() failed with ret=0x000000F0). 

Some idea?

        ```
ATCAIfaceCfg *gCfg = &cfg_ateccx08a_i2c_default;
        ATCA_STATUS status = ATCA_GEN_FAIL;
        uint8_t random_number[32];
        uint8_t serial_number[9];

        gCfg->iface_type = ATCA_I2C_IFACE,
        gCfg->devtype = ATECC608A,
        gCfg->atcai2c.slave_address = TAU_I2C_ADDRESS_SECURITY;
        gCfg->atcai2c.bus = 1;
        gCfg->atcai2c.baud = 100000;
        gCfg->wake_delay = 800;
        gCfg->rx_retries = 20;

        status = atcab_init(gCfg);

        if (status == ATCA_SUCCESS)
        {
            status = atcab_random(random_number);
            ESP_LOGV(DEV_TAG, "\tRandom: %i", status);

            status = atcab_read_serial_number(serial_number);
            ESP_LOGV(DEV_TAG, "\tSerial: %i", status);

            status = atcab_release();
            ESP_LOGV(DEV_TAG, "\tRelease: %i", status);
        }
        else
        {
            ESP_LOGE(DEV_TAG, "\t- ERROR: %i", status);
        }
```

JavierPascual, 2019-05-23:
> We don't currently have an I2C driver written for the ESP32. We are aware of the desirability of an ESP32 integration. Do you have a Microchip FAE contact already? If you do you can send them a [...]

Hi @bryan-hunt! Could you tell me who is the person to contact about ESP32 I2C interface? Thanks a lot!!

PBearson, 2019-09-19:
Hi, I have tried re-implementing @bryan-hunt's solution on ESP-IDF release/v4.0, but now I get a compilation error stating that "cryptoauthlib.h" is not recognized. Has something changed about the way ESP-IDF builds external components? I never had this problem before.

bryan-hunt, 2019-09-19:
@PBearson I will have to assume something has changed then in the ESP build system.

PBearson, 2019-09-20:
I managed to solve the issue by placing cryptoauthlib into a new components directory, which is at the root of the project.

bugabuga64, 2019-10-25:
@PBearson , did you manage to complete the integration of the ESP32 with ATECC608A ? I am looking for such project, could you share the link here ?

PBearson, 2019-10-25:
@bugabuga64 Yes, I managed to integrate them. See here: https://github.com/PBearson/eccx08-with-esp32