MicrochipTech/cryptoauthlib #6: big-endian

uzsunliming1, 2018-04-01:
Hi,
what i want to know is that if the lib could work in a big-endian system......

bryan-hunt, 2018-04-02:
We have not built and tested the library on a big endian machine. But we do know that some of the functions (namely the host side hash and MAC calculations) do not have endianness considerations in their source so will most likely not work properly. However most of the library is communications related which are simple byte arrays where endianness should not be an issue.

* What are your use cases you're looking to use the devices for?
* What is your target's actual architecture and compiler?

uzsunliming1, 2018-06-05:
for all the big-endian system:
1.atca_command.h: struct "ATCAPacket",member of :uint16_t param2;--->when forcing the ATCAPacket into a bytes stream  sending on I2c or SWI bus ,it is a problem;
2.atca_command.c: void atCRC(size_t length, const uint8_t *data, uint8_t *crc_le),only work with little-endian system;

the other codes still need to be check......

watersb1323, 2018-11-06:
I have managed to perform most operations on my Big Endian chip by reversing the order of param2 within the atCalcCrc function within atca_command.c.

@bryan-hunt Would you be able to shed any light on why the atca_priv_write function may be failing for a big endian system?

BenUdall-Microchip, 2018-11-06:
Hi @watersb1323 , I can't say for sure, but it's worth noting that atcab_priv_write() can perform an unecrypted or encrypted PrivWrite command depending on whether a write_key is provided. Unencrypted PrivWrite commands (write_key=NULL) can only be performed when the config zone is locked and the data zone is unlocked. After the data zone is locked, only encrypted PrviWrite commands are allowed (if enabled by SlotConfig bit 14).

An encrypted PrivWrite makes use of the host functions (atcah_), which will use the [host SHA256 functions](https://github.com/MicrochipTech/cryptoauthlib/blob/master/lib/crypto/hashes/sha2_routines.c) to calculated appropriate keys and MACs. We suspect the SHA256 function would need to be updated to work on a big-endian system.

watersb1323, 2018-11-06:
Thanks @bryan-hunt, that makes sense alright. I have performed unencrypted write and thus both config and data zones have been locked. My error is coming as a result of the trying to run the provision command again, i.e. performing an encrypted write.

Currently I am investigating your source code for the required change necessary to allow this code to run on a Big endian system. I am focusing on where the variables key_id and write_key_id are being passed and subsequently used in functions within atcab_priv_write(). It appears they are used heavily in the host functions (atcah_gen_dig() & atcah_privwrite_auth_mac) as you suggested (mostly bit manipulations). I presume this will only be a minor change to one of those two key id's as far as I can see. Would you have any suggestions as to what might be required as I ave been looking at this for some time and am at a standstill?

BenUdall-Microchip, 2018-11-06:
[atcah_gen_dig()](https://github.com/MicrochipTech/cryptoauthlib/blob/c6b176e7dfd8a74db7869995187faf92f8fc8abc/lib/host/atca_host.c#L748) and [atcah_privwrite_auth_mac()](https://github.com/MicrochipTech/cryptoauthlib/blob/c6b176e7dfd8a74db7869995187faf92f8fc8abc/lib/host/atca_host.c#L1010) are appropriately explicit with handling the endianess of key_id, so I don't think the problem will be there.
They both rely on atcac_sw_sha2_256(), which ends up in the SHA256 implementation in [lib/crypto/hashes/sha2_routines.c](https://github.com/MicrochipTech/cryptoauthlib/blob/master/lib/crypto/hashes/sha2_routines.c), which doesn't handle big-endian systems. Unfortunately, this probably isn't a trivial update.

One can make use of the crypto unit tests in [test/atca_crypto_sw_tests.c](https://github.com/MicrochipTech/cryptoauthlib/blob/master/test/atca_crypto_sw_tests.c) to validate the software SHA1 and SHA256 implementations.

uzsunliming1, 2019-10-02:
I don't have a big-endian on hand,but i think it will work.....thanks for the work.

closed 2019-10-02 by uzsunliming1