MicrochipTech/cryptoauthlib #9: Potential bug in atcab_secureboot_mac()

deangereaux, 2018-05-14:
This may be intentional, however in the atcab_secureboot_mac() function in atca_basic_secureboot.c source file

On line #168 the current code is:
      // Encrypt the digest
      memcpy(&sboot_enc_params, digest, sizeof(sboot_enc_params));

Is the memcpy() call correct?  Shouldn't this be memset() instead? Like this:
     memset(&sboot_enc_params, 0, sizeof(sboot_enc_params));

Dean Gereaux
Golden Bits Softare

BenUdall-Microchip, 2018-05-14:
Hi Dean,
Good catch! You're absolutely right, that should be memset(). The bug results in a buffer over-read of the digest buffer.

I've logged the issue and will have it fixed in the next release.

bryan-hunt, 2018-08-30:
Fixed in beac5a2 (Release 5 - 20180718)

closed 2018-08-30 by bryan-hunt