MicrochipTech/cryptoauthlib #14: possible bug in atca_execution.c

uzsunliming1, 2018-06-05:
atca_execute_command():
.........................................................
       do
        {
            // receive the response
            if ((status = atreceive(device->mIface, packet->data, &(packet->rxsize))) == ATCA_SUCCESS)
            {
                break;
            }

#ifndef ATCA_NO_POLL
            // delay for polling frequency time
            atca_delay_ms(ATCA_POLLING_FREQUENCY_TIME_MSEC);
#endif
        }
        while (max_delay_count-- > 0);
////////////////////////////////////////////////////////////////////
//i think below lines should be added to check status:
		if ( status != ATCA_SUCCESS)
		{
			break;
		}
///////////////////////////////////////////////////////////////////
// Check response size
		if (packet->rxsize < 4)
		{
			if (packet->rxsize > 0)
			{
				status = ATCA_RX_FAIL;
			}
			else
			{
				status = ATCA_RX_NO_RESPONSE;
			}
			break;
		}
..................................................................................................

uzsunliming1, 2018-06-05:
besides,regarding command completion polling,why we couldn't delay a typical time first befor start polling?
rather than polling immediately after command sending? 
this was also mentioned in datasheet,section 6.2.2;
delay a typical time first could save polling times......offloading the CPU and the chip.

uzsunliming1, 2018-06-06:
besides,we could try a delay of typical time tEXEC (typical) before polling,rather than immediately start polling after command sending,this could offload the CPU including the chip;
this was also mentioned in datasheet section 6.2.2(atsha).

BenUdall-Microchip, 2018-06-06:
Thanks for the reports! I'm not sure the wait typical time change will make it into the next release, but your other points are good and will be there.

uzsunliming1, 2018-06-07:
another question:
pointless to Check response size in  atca_execute_command():
...............................................................................................................
// Check response size
        if (packet->rxsize < 4)
        {
            if (packet->rxsize > 0)
            {
                status = ATCA_RX_FAIL;
            }
            else
            {
                status = ATCA_RX_NO_RESPONSE;
            }
            break;
        }
...............................................................................................................
in function atreceive(),rxlength is passed by pointer,which means it will be updated(as a return value),
in fact,actual fuctions such as hal_i2c_receive(){in hal_i2c_bitbang.c} and hal_swi_receive(){in hal_swi_bitbang.c}  didn't update rxlength at all,it updated rxdata[0] as actual length of data bytes been read;
so,revising atreceive(), just pass packet->rxsize by value rather than by pointer when calling atreceive(); which will only be used as expected reading length.

uzsunliming1, 2018-06-07:
actual fuctions such as hal_i2c_receive(){in hal_i2c_bitbang.c} and hal_swi_receive(){in hal_swi_bitbang.c} ，they were also polling the chip(retries),then does that mean polling is duplicated if ATCA_NO_POLL was not defined?

bryan-hunt, 2018-08-30:
Fixed in 8ed4303 (Release 5 - 20180718)

closed 2018-08-30 by bryan-hunt