MicrochipTech/cryptoauthlib #24: Arduino support

gyengus, 2018-08-09:
Do you have, or plan arduino support?

ronnytittoto, 2018-08-14:
Check these projects, would they help?
https://github.com/arduino-libraries/ArduinoBearSSL
https://github.com/arduino-libraries/ArduinoECCX08
https://github.com/CryptoThings/AtCryptoAuthLib

Misiu, 2019-01-18:
@bryan-hunt it would be awesome to have an official library.
I'm developing using https://github.com/esp8266/Arduino so Arduino/ESP8266/ESP32 support would be awesome!
https://github.com/CryptoThings/AtCryptoAuthLib is a good starting point

GabrielNotman, 2019-01-22:
@gyengus @Misiu I have been working on fork here: https://github.com/GabrielNotman/cryptoauthlib

I still need to add some instructions and examples but here are the main points:

- The library needs to be installed in one of the Arduino library folders (not the sketch folder).
- Currently only implements i2c and i2c bitbang.
- Arduino HAL selection is done via /src/arduino_hal_select.h
- For i2c, bus 0 must be selected (multi-bus support is not supported yet).

I've tested it with a SAMD21 based board. For AVR boards you might need to use bitbang due to _Wire_ buffer sizes.

gannaramu, 2020-11-08:
@GabrielNotman any plans of syncing it up with the latest release from Microchip? I was looking to use with the Atecc Trust and Go chips ATECC608 TNGTLS.

GabrielNotman, 2020-11-15:
@gannaramu I'm currently working on updating it to v3.2.4.

gannaramu, 2020-11-15:
@GabrielNotman That would be really helpful!. I can help you test things I have a cryptoauth trust platform connected to a teensy4.0 board. For the past couple of days, I have been modifying the https://github.com/sparkfun/SparkFun_ATECCX08a_Arduino_Library to be able to read the X509 certificate stored in the device. I was still not sure how to decompress the certificate to a PEM format. Could you please guide me with any relevant resources?.

Thanks
Ram Rohit