MicrochipTech/cryptoauthlib #27: Creating device CSR, how to get the certificate definition i.e atcacert_def_t

umanayana, 2018-08-30:
I am trying to create a the CSR for the device so I can create the device certificate. In order to create the CSR I need to create the CSR template (i. e `atcacert_def_t`). To create the CSR template I am looking at the python script `cert2certdef.py` which takes a CSR using `--device-csr`.  

The `CSR` I feed the script is a CSR that I generate using openSSL with specific certificate element such as origination name, etc. The `CSR` of obviously not correct CSR for the device because the key pair used to create it is wrong. But the common elements would be correct.

Is this the correct way to get the CSR definition template?

umanayana, 2018-08-31:
Looks like asked this question in the wrong repository. Please close I asked it in the correct repo
https://github.com/MicrochipTech/aws-iot-zero-touch-secure-provisioning-kit

closed 2018-11-06 by bryan-hunt

DavidZemon, 2019-06-25:
Issue is still open without any responses. I'm struggling to understand all this certificate jargon as well.

https://github.com/MicrochipTech/aws-iot-zero-touch-secure-provisioning-kit/issues/12