MicrochipTech/cryptoauthlib #31: atcab_genkey and atcacert_read_cert return  ATCA_COMM_FAIL on raspberry with i2c

radfm, 2018-09-06:
Hello,
I have problem with some commands when I use ATECC508 on raspberry with i2c. The same code works fine if I run it with ATECC on AT88CK101. Also other commands ( atcab_priv_write, CheckLockStatus, etc.) work fine on raspberry.

Commands atcab_genkey and  atcacert_read_cert return ATCA_COMM_FAIL.

The problem arises during reading response from /dev/i2c-1

radfm, 2018-09-06:
Now I download and use latest version of cryptoauthlib. 

Errorcode is now different -  F3 - ATCA_WAKE_SUCCESS

Can you tell me, what it means in case of atcab_genkey or atcacert_read_cert commands?

Hikari1026, 2021-06-07:
Hello everyone, sorry if I step in like this, but I'm facing the same issue, under slightly different cirumstances. I'm trying to make the ATECC chips work with Arduino Uno via I2C, and so far it worked with every command, except for atcab_genkey and the signing related ones as well.

Basically, if I connect the ATECC608A it returns error code 0xF3, while if I try to use the ATECC608B it just stalls out forever (like it's expecting the data from the device but it won't respond to it). I know for a fact that both chips are working properly (tested them with a SAM D21 Xplained Pro), so I was wondering if you managed to find a fix.

Regardless, thank you for your time.

bryan-hunt, 2021-06-21:
I'm leaving this note for those coming across this issue in the future.

These questions come up a lot and invariably they are configuration or hardware problems - the UDFN package is sometimes difficult to solder and many of the commands require very little power while some commands that require many cycles of the random number generator will cause cold solder joints to present themselves. Regardless they are product support questions which we handle through the Microchip Support Portal.

For product support please submit a request through the [mySupport portal](https://microchipsupport.force.com/s/) where Microchip's global team of FAEs can work on your use case and questions.

For details on the mySupport Portal or how to use it please see the [instructions for using the mySupport portal](https://microchipsupport.force.com/s/article/How-to-submit-a-case)

closed 2021-06-21 by bryan-hunt