MicrochipTech/cryptoauthlib #33: PrivWrite command can't write unencrypted data when use Python

explora26, 2018-09-10:
I'm trying to write a  unencrypted private key  to the ATECC508A with python API, but I always get the return code "ATCA_EXECUTION_ERROR".

I think the issuse is that "write_key" is not NULL, maybe we can fix it by following:

    def atcab_priv_write(key_id, priv_key, write_key_id, write_key):
        if write_key:
            status = _CRYPTO_LIB.atcab_priv_write(key_id, bytes(priv_key), write_key_id, bytes(write_key))
        else:
             status = _CRYPTO_LIB.atcab_priv_write(key_id, bytes(priv_key), write_key_id, None)
        return status

bryan-hunt, 2018-09-10:
This is a use case that should __never__ be used. The device only supports unencrypted priv_write commands when the data zone is unlocked. When the datazone is locked regardless of the slot configuration all unencrypted priv_write commands are rejected.

closed 2018-10-26 by bryan-hunt

conroy-c, 2023-12-13:
@bryan-hunt but what if we're trying to manually provision devices before the data zone is locked? We just ran into this today trying to provision some test devices for our project with known test private keys.