MicrochipTech/cryptoauthlib #36: Help to install with pip install

NSLog0, 2018-09-22:
I ran pip install then I got an error. How should I fix it
[image]

NSLog0, 2018-09-22:
Thank you, For you supporting. 

It looks like an option that you prefer didn't work

NSLog0, 2018-10-05:
@bryan-hunt Hi, I'm still waiting for your support, if you have the alternative way to use ATEC chip with python, can you suggest while you're working with an issue

NSLog0, 2018-10-16:
@BenUdall-Microchip I'm haven't tried a new version yet, but I tried your current version with RPi model 3 is not working too. I will capture error after I on may laptop

NSLog0, 2018-10-16:
@BenUdall-Microchip Have you update a version yet?, Please remind me after you finished

bryan-hunt, 2018-11-01:
@NSLog0 A new version was released last week that should resolve most issues. Make sure to use the `--upgrade` option when using pip.

closed 2018-11-14 by bryan-hunt