MicrochipTech/cryptoauthlib #37: atcab_generate_csr generates public key from the wrong slot

mindcruzer, 2018-10-20:
Hey there,

`atcab_generate_csr()` can either retrieve the public key from a slot on the device, or generate it from a private key. When `is_genkey` is requested for the source of the public key the wrong slot number is passed into `atcab_get_pubkey()`.

https://github.com/MicrochipTech/cryptoauthlib/blob/979f099692c1843609dbc51e6127c6c86003262d/lib/atcacert/atcacert_client.c#L259

`key_slot` is the slot for the public key, which in this instance is effectively undefined. This line should read:

```
status = atcab_get_pubkey(priv_key_slot, pub_key);
```

BenUdall-Microchip, 2018-10-22:
Hi @mindcruzer, this is the results of a little over-specification in the [atcacert_def_t](https://github.com/MicrochipTech/cryptoauthlib/blob/979f099692c1843609dbc51e6127c6c86003262d/lib/atcacert/atcacert_def.h#L149) structure. When ```atcacert_def_t.public_key_dev_loc.is_genkey``` is true, then ```atcacert_def_t.public_key_dev_loc.slot``` should have the slot number for the private key, which should be the same as the ```atcacert_def_t.private_key_slot``` value.

mindcruzer, 2018-10-22:
Ok thanks for clarifying @BenUdall-Microchip

closed 2018-10-22 by mindcruzer