MicrochipTech/cryptoauthlib #40: Cmake cleanup for ATCA_HAL_I2C target

balaji-reddy, 2018-12-05:
Hello,

I've tried to cross compile the cryptoauthlib for arm architecture with target ATCA_HAL_I2C and found some kinds of stuff are not required for that target. Attached the patch for all changes.

[cryptoauthlib_ATCA_HAL_I2C_issues.patch.gz](https://github.com/MicrochipTech/cryptoauthlib/files/2650192/cryptoauthlib_ATCA_HAL_I2C_issues.patch.gz)

bryan-hunt, 2018-12-05:
This breaks python and pkcs11 builds that include both i2c & hid support.

Mostly it looks like you're seeing that when you don't include hid support it is still attempting to link against usb libraries. That we can fix.

nknotts, 2018-12-21:
Along the same lines, I would appreciate not linking against udev if compiling without hid.

bryan-hunt, 2019-03-04:
This should be resolved in the latest release (20190304) in f86b9db with better hal options and dependency inclusion.

closed 2019-03-07 by bryan-hunt