MicrochipTech/cryptoauthlib #42: clarification on atcab_write_enc / atcab_read_enc commands

syedelec, 2018-12-12:
Hi,

I would like to understand why atcab_write_enc and atcab_read_enc take the key and key slot used to encrypt / decrypt the message in parameter ?

In that case, any application that uses these functions is vulnerable no ? since the encryption key needs to be given in plain text.

Is it possible to only specify the encryption key slot in order to perform an atcab_(write/read)_enc ?

bryan-hunt, 2018-12-12:
Regardless of your system cryptographic keys have to be in ram somewhere in order to perform cryptographic operations with them.

The slot is the slot you're reading/writing to and the configuration of the device determines the key the device will use to perform those operations. On the host/mcu it has to be able to also perform the other operation so the encrypted read and write commands accept the key as an input.

syedelec, 2018-12-12:
ok thank you for the details

closed 2018-12-12 by syedelec