MicrochipTech/cryptoauthlib #46: possible silicon bug in swi version(SOT23-3)

uzsunliming1, 2019-01-15:
for ATSHA204A,i tested UpdateExtra command on some chips in SOT23-3 footprint,below is a description of  the potiential problem:
1. config&data&otp keep unlocked,unchanged, as it was shipped:
  Selector Mode ==0;
  UserExtra == 0;
  Selector == 0;
  Limit Use for slot 5 is true;Useflag 5 == 0xFF;
2.chip A:run UpdateExtra with UPDATE_MODE_DEC_COUNTER,NewValue = 5;
             then check: UserExtra == 5; (??)
                               Useflag 5 == 0xFF; (no change)  
3,Chip B:run 20+ times of UpdateExtra with UPDATE_MODE_DEC_COUNTER,NewValue = 0:
             for the first 10+ times,chip return success; 
             after 20+ times run the UpdateExtra command,chip return atca_status == 0xf4;        
                                      then check:UserExtra == non zero; (??)
i also tested another chip with same footprint,but config&data area was locked,and UserExtra == non zero;when run UpdateExtra with UPDATE_MODE_DEC_COUNTER,NewValue = 5;(while Limit Use for slot 5 is true;Useflag 5 == 0x3F;),but chip always return an err code f4,Useflag 5 keep unchanged.

Does that mean if  UserExtra == non zero,UpdateExtra with UPDATE_MODE_DEC_COUNTER will always fail for SWI version?(TWI version tested OK in same envitonment).
i will test more and give feedback.

bryan-hunt, 2019-01-15:
Please submit product support requests through [myMicrochip](https://www.microchip.com/mymicrochip/)

uzsunliming1, 2019-01-18:
today,i tested one chip(sot23-3) from the other sourcing,everything tested OK,i found that it's revision number is :00 02 00 09;
and the revision of the chips mentioned in my initial post read is :00 00 00 04,i believe those chips are of older version(i think they are ATSHA204,without 'A');and revison 00 02 00 09 means ATSHA204A(for both swi--sot23 version and twi--soic8 version).
as mentioned in new datasheet(9. Compatibility),UpdateExtra with UPDATE_MODE_DEC_COUNTER was only supported by ATSHA204A.
all my tests only confirmed the difference between older chip and new ones;
i will close this very soon.