MicrochipTech/cryptoauthlib #47: Repository contains CRLF line endings

vgottardi, 2019-01-16:
The CRLF line endings create issues when working in Linux and OS X.
https://help.github.com/articles/dealing-with-line-endings/

bryan-hunt, 2019-01-16:
Yes, some of our style enforcement tools need to be updated. While we do currently build on travis.ci for linux and osx it is a specialized use case. Which tool are you experiencing trouble with?

vgottardi, 2019-01-16:
I'm getting issues making a yocto patch:
patching file lib/hal/hal_linux_i2c_userspace.c
Hunk #1 FAILED at 40 (different line endings).

bryan-hunt, 2019-03-04:
This should be resolved by f86b9db

vgottardi, 2019-03-05:
Great, however I'm using the pkcs11 branch. Will there be a merge soon (from master to pkcs11 or the other way around)?

bryan-hunt, 2020-12-03:
This should have been resolved with the mainlining of pkcs11. I'm closing this ticket as being stale.

closed 2020-12-03 by bryan-hunt