MicrochipTech/cryptoauthlib #49: Is there a typo in atca_tests_gendig.c?

habebit, 2019-01-24:
Hi there, thanks for your work. Is there a typo? I thought the configuration zone should be locked, before you can generate a private key and a corresponding public key. I think it should be called `...if data zone is` **locked** instead of ...
https://github.com/MicrochipTech/cryptoauthlib/blob/2044ad999eb288ef9c70b52bd15d54464a0a6a3c/test/atca_tests_genkey.c#L36

bryan-hunt, 2019-01-24:
The line directly above the one listed says the configuration zone should be locked:

```
/** \brief this test assumes a specific configuration and locked config zone
 * test will generate a private key if data zone is unlocked and return a public key
 * test will generate a public key based on the private key if data zone is locked
 */
```
Genkey can also be run when the data zone is unlocked (and would be a necessity when the configuration doesn't allow a genkey on a slot after the device is provisioned)

habebit, 2019-01-24:
Sorry, you are absolutly right. My mistake! Thanks anyway.

closed 2019-01-24 by bryan-hunt