MicrochipTech/cryptoauthlib #50: Setting up cryptoauthlib as a PKCS11 Provider - module failed to initialize

matriedel, 2019-01-26:
I was following the manual "Setting up cryptoauthlib as a PKCS11 Provider"
I got stuck with the following error

sudo p11tool --provider=/usr/lib/libcryptoauth.so
p11-kit: (unknown): module failed to initialize: Internal error
pkcs11_add_provider: PKCS #11 error.

One guess was the issue might be caused by a wrong slot config file. I defined 
/var/lib/cryptoauthlib/slot.conf as
----------------------------------------
interface = i2c,0xB0
# freeslots = 1,2,3

# Slot 0 is the primary private key
object = private,device,0
---------------------------------------

what I was missing in the config file was a bus number like e.g.
atcai2c.bus = 0

Where can I configure the bus setting?
Any other hint of the error I see?

matriedel, 2019-01-26:
With adding your proposed changes the issue could be solved.

But with further p11tool --export-pubkey testing, I get another issue "Error in pkcs11_export_pubkey:821:"

p11tool --export-pubkey "pkcs11:token=0123EE;object=device;type=private"
warning: --login was not specified and it may be required for this operation.
warning: no --outfile was specified and the public key will be printed on screen.
Error in pkcs11_export_pubkey:821: The requested PKCS #11 object is not available

What might be the issue here?

bryan-hunt, 2019-01-26:
What happens when you do a list-all or list-tokens?

There are some troubleshooting tips on this page: https://github.com/MicrochipTech/cryptoauthlib/wiki/Greengrass-HSI

matriedel, 2019-01-26:
For p11tool --list-all I  get a long list with 1718 Objects
for p11tool --list-tokens I get the following
Token 0:
        URL: pkcs11:model=p11-kit-trust;manufacturer=PKCS%2311%20Kit;serial=1;token=System%20Trust
        Label: System Trust
        Type: Trust module
        Manufacturer: PKCS#11 Kit
        Model: p11-kit-trust
        Serial: 1
        Module: p11-kit-trust.so

bryan-hunt, 2019-01-26:
If you’re not using p11-kit you have to specify the provider with every command. So list-all and list-tokens with the provider set produces?

matriedel, 2019-01-26:
O.K I see. Thank you for your comment.

Unfortunately, it seems I cannot connect to the device anymore via p11tool.
The error message is slightly different than at the first time.

p11tool --provider=/usr/lib/libcryptoauth.so
p11-kit: (unknown): module failed to initialize: An error occurred on the device
pkcs11_add_provider: PKCS #11 error in device

What I did to get a clean start I was running successfully 
\cryptoauthtools\python\examples\config.py
At least this indicates that a communication with the device is possible.

bryan-hunt, 2019-01-26:
model=p11-kit-trust;manufacturer=PKCS%2311%20Kit;serial=1;token=System%20Trust is a software token from p11-kit and does not indicate that p11tool was able to communicate with the cryptoauth device.

See https://github.com/MicrochipTech/cryptoauthlib/wiki/PKCS11-Linux-Setup#using-p11-kit-proxy for how to set up p11-kit to see the library and interact with it.

Did you change the bus number in [atca_cfgs.c](https://github.com/MicrochipTech/cryptoauthlib/blob/21f9f26f3dae7194177f8be1b38bf8081616cd61/lib/atca_cfgs.c#L45) for your platform, rebuild, and reinstall the .so?

matriedel, 2019-01-26:
Yes I changed the bus number to .atcai2c.bus = 0 and .devtype = ATECC508A, in atca_cfgs.c for my platform, rebuild, and reinstalled the .so
After I did it the first time 
p11tool --provider=/usr/lib/libcryptoauth.so
worked one time.
Quite strange....

matriedel, 2019-01-31:
After cleaning up everything and reinstallation it works.

matriedel, 2019-01-31:
Problem solved after reinstallation and modification of atca_cfgs.c

closed 2019-01-31 by matriedel