MicrochipTech/cryptoauthlib #52: Create a CSR for the private key fails

matriedel, 2019-01-31:
All tests listed are successful e.g. "Get the public key for a private key"

sudo p11tool  --export-pubkey "pkcs11:token=0123EE;object=device;type=private"
warning: --login was not specified and it may be required for this operation.
warning: no --outfile was specified and the public key will be printed on screen.
-----BEGIN PUBLIC KEY-----
MFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAELw702nQwPHTtqllf+pGlAvMTg1KP
uvaJh2ansPrizzdZw+TIZRo72zzl7lIVI8lNdgl514wlXfv8soaB3D0lSw==
-----END PUBLIC KEY-----

except for the test "Create a CSR for the private key"

sudo openssl req -engine pkcs11 -key "pkcs11:token=0123EE;object=device;type=private" -keyform engine -new -out new_device.csr -subj "/CN=NEW CSR EXAMPLE"
engine "pkcs11" set.
Found slot without user PIN
PKCS11_get_private_key returned NULL
cannot load Private Key from engine
3070100688:error:26096080:engine routines:ENGINE_load_private_key:failed loading private key:eng_pkey.c:124:
unable to load Private Key

Do you have any hint for the root cause?

bryan-hunt, 2019-01-31:
The best way to diagnose that is to enable logging in the p11-kit .module configuration

bryan-hunt, 2019-02-01:
Well that's a good indication that there is a problem with your OpenSSL configuration or install (either openssl, libp11, or p11-kit) since it's not making any calls through p11-kit.

bryan-hunt, 2019-02-01:
Yes something in the openssl configuration (wrong libp11 being used) or libp11 (not finding p11-kit) is the problem. If you have entries in openssl.cnf that are configuring the pkcs11 engine you should revert the file back to default and try.

Other than that I don't have anything specific to diagnose as it is dependent on your system. The steps described in the [wiki](https://github.com/MicrochipTech/cryptoauthlib/wiki/PKCS11-Linux-Setup) do work with new clean installs of Ubuntu (on a PC or VM) and Raspbian.

matriedel, 2019-02-01:
Yes, you were right - there were some entries in the openssl.cnf. After removing them I see a different error message.

----------------------------
sudo openssl req -engine pkcs11 -key "pkcs11:token=0123EE;object=device;type=private" -keyform engine -new -out new_device.csr -subj "/CN=NEW CSR EXAMPLE"

Unable to load module (null)
can't use that engine
3069662416:error:81065401:libp11:pkcs11_CTX_load:Unable to load PKCS#11 module:p11_load.c:79:
3069662416:error:260B806D:engine routines:ENGINE_TABLE_REGISTER:init failed:eng_table.c:175:
no engine specified
unable to load Private Key
-----------------------------

A link to the module /usr/lib/libcryptoauth.so
is now only defined in 
/usr/share/p11-kit/modules/cryptoauthlib.module

module: /usr/lib/libcryptoauth.so
critical: yes
trust-policy: yes
managed: yes
log-calls: yes

bryan-hunt, 2019-02-01:
That means the pkcs11 isn't installed in the system where openssl is expecting it. See [libp11](https://github.com/OpenSC/libp11). You could try using the configuration you had before but not specifying MODULE_PATH to see if libp11 is able to located p11-kit

matriedel, 2019-02-01:
not specifying MODULE_PATH seems not to work

Error configuring OpenSSL
3069564112:error:25066067:DSO support routines:DLFCN_LOAD:could not load the shared library:dso_dlfcn.c:187:filename(/usr/lib/ssl/engines/libpkcs11.so): /usr/lib/ssl/engines/libpkcs11.so: cannot open shared object file: No such file or directory
3069564112:error:25070067:DSO support routines:DSO_load:could not load the shared library:dso_lib.c:233:
3069564112:error:260B6084:engine routines:DYNAMIC_LOAD:dso not found:eng_dyn.c:467:
3069564112:error:260BC066:engine routines:INT_ENGINE_CONFIGURE:engine configuration error:eng_cnf.c:191:section=pkcs11_section, name=dynamic_path, value=/usr/lib/ssl/engines/libpkcs11.so
3069564112:error:0E07606D:configuration file routines:MODULE_RUN:module initialization error:conf_mod.c:223:module=engines, value=engine_section, retcode=-1

bryan-hunt, 2019-02-01:
Definitely consult with the libp11 documentation to troubleshoot where its supposed to be on your system.

matriedel, 2019-02-01:
O.K thanks

matriedel, 2019-02-07:
I further analysed the issue. Giving explicitly all path information it seems to work

sudo openssl engine dynamic -pre SO_PATH:/usr/local/lib/engines-1.1/pkcs11.so -pre ID:pkcs11 -pre LIST_ADD:1 -pre LOAD -pre MODULE_PATH:/usr/lib/libcryptoauth.so

_(dynamic) Dynamic engine loading support
[Success]: SO_PATH:/usr/local/lib/engines-1.1/pkcs11.so
[Success]: ID:pkcs11
[Success]: LIST_ADD:1
[Success]: LOAD
[Success]: MODULE_PATH:/usr/lib/libcryptoauth.so
Loaded: (pkcs11) pkcs11 engine_

Checking what links are used
openssl version -a

OPENSSLDIR: "/usr/local/ssl"
ENGINESDIR: "/usr/local/lib/engines-1.1"

I modified /usr/lib/ssl/openssl.cnf
with 
....
_dynamic_path = /usr/local/lib/engines-1.1/libpkcs11.so 
MODULE_PATH = /usr/lib/libcryptoauth.so_
...

But the setting in openssl.cnf seems to have no effect as a wrong path /opt/lib/engines-1.1/ is selected.
This is the reason why the following command fails

openssl req -engine pkcs11 -key "pkcs11:token=0123EE;object=device;type=private" -keyform engine -new -out new_device.csr -subj "/CN=NEW CSR EXAMPLE"

_invalid engine "pkcs11"
3069222912:error:25066067:DSO support routines:dlfcn_load:could not load the shared library:crypto/dso/dso_dlfcn.c:117:filename(/opt/lib/engines-1.1/pkcs11.so): /opt/lib/engines-1.1/pkcs11.so: cannot open shared object file: No such file or directory_

Do you have any hint why settings in openssl.cnf have no effect?

syedelec, 2019-02-12:
@matriedel try [look at this link](https://unix.stackexchange.com/questions/279415/openssl-does-not-load-engine-from-config-file/) to have `openssl.cnf` taken into account

basically you need to export `OPENSSL_CONF` variable:

`export OPENSSL_CONF=/path/to/config`

matriedel, 2019-02-13:
O.K got it . Thank you a lot for your help

closed 2019-02-13 by matriedel

hirenjenex, 2022-03-10:
I am getting the same error can you please help to resolve it.

```
revofi@revofi-desktop:~/cryptoauthlib$ sudo openssl req -engine pkcs11 -key "pkcs11:token=00ABC;object=device;type=private" -keyform engine -new -out new_device.csr -subj "/CN=NEW CSR EXAMPLE"
[sudo] password for revofi:
engine "pkcs11" set.
C_Initialize
  IN: pInitArgs = NULL
C_Initialize = CKR_OK
C_GetSlotList
  IN: tokenPresent = CK_FALSE
  IN: pulCount = 0x7FFAA81E10 = 1
 OUT: pSlotList = (1) NO-VALUES
C_GetSlotList = CKR_OK
C_GetSlotList
  IN: tokenPresent = CK_FALSE
  IN: pulCount = 0x7FFAA81E10 = 1
 OUT: pSlotList = (1) [ SL0 ]
C_GetSlotList = CKR_OK
C_GetSlotInfo
  IN: slotID = SL0
 OUT: pInfo = {
        slotDescription: "0_3_0"
        manufacturerID: "Microchip Technology Inc"
        flags: 5 = CKF_TOKEN_PRESENT | CKF_HW_SLOT
        hardwareVersion: 0.3
        firmwareVersion: 255.255
      }
C_GetSlotInfo = CKR_OK
C_GetTokenInfo
  IN: slotID = SL0
 OUT: pInfo = {
        label: "00ABC"
        manufacturerID: "Microchip Technology Inc"
        model: "ATECC608B"
        serialNumber: "23B6E125B96D5901"
        flags: 1025 = CKF_RNG | CKF_TOKEN_INITIALIZED
        ulMaxSessionCount: 1
        ulSessionCount: 0
        ulMaxRwSessionCount: 1
        ulRwSessionCount: 0
        ulMaxPinLen: 128
        ulMinPinLen: 0
        ulTotalPublicMemory: 1
        ulFreePublicMemory: 1
        ulTotalPrivateMemory: 1
        ulFreePrivateMemory: 1
        ulFreePrivateMemory: 1
        hardwareVersion: 0.3
        firmwareVersion: 255.255
        utcTime:
      }
C_GetTokenInfo = CKR_OK
Found slot without user PIN
Found slot without user PIN
PKCS11_get_private_key returned NULL
cannot load Private Key from engine
548155942528:error:80067065:pkcs11 engine:ctx_load_privkey:object not found:eng_back.c:876:
548155942528:error:26096080:engine routines:ENGINE_load_private_key:failed loading private key:../crypto/engine/eng_pkey.c:78:
unable to load Private Key

```