MicrochipTech/cryptoauthlib #53: Rights under Raspbian? Adding user to i2c group is not enough

i7clock, 2019-02-12:
My application works under root rights but does not with my created user. I added the user to the i2c group but that didn't do the trick. What rights does the library need?

I get f0 as return. Probably returned by the function hal_i2c_init() in hal_linux_i2c_userspace.c.

bryan-hunt, 2019-02-12:
Probably would be easiest if you configured udev rules:

https://www.raspberrypi.org/forums/viewtopic.php?t=26212

i7clock, 2019-02-12:
Thanks! This din't help in my case, but I found this which also delas with the udev rules:

[http://www.netzmafia.de/skripten/hardware/RasPi/RasPi_I2C.html](http://www.netzmafia.de/skripten/hardware/RasPi/RasPi_I2C.html)

closed 2019-02-12 by i7clock