MicrochipTech/cryptoauthlib #54: Custom HAL using Python

tjstone14, 2019-02-14:
I was looking into using Python to with a custom HAL.  I have done a similar thing in C, but have not been able to get Python to work.  Is this possible?

I have attached my code here.  It always crashes in` atcab_init()` before any of the HAL callbacks are called.

```
from cryptoauthlib import *
import ctypes as c

ATCA_SUCCESS = 0x00
revision = bytearray(4)
randomnum = bytearray(32)

class Ecc:
    def __init__(self):
        load_cryptoauthlib()
        icfg = ATCAIfaceCfg()
        icfg.devtype = 2
        icfg.iface_type = 5
        icfg.cfg.atcacustom.halinit = c.cast( self.halinit, c.c_void_p)
        icfg.cfg.atcacustom.halpostinit = c.cast( self.halpostinit, c.c_void_p)
        icfg.cfg.atcacustom.halsend = c.cast( self.halsend, c.c_void_p)
        icfg.cfg.atcacustom.halreceive = c.cast( self.halreceive, c.c_void_p)
        icfg.cfg.atcacustom.halwake = c.cast( self.halwake, c.c_void_p)
        icfg.cfg.atcacustom.halidle = c.cast( self.halidle, c.c_void_p)
        icfg.cfg.atcacustom.halsleep = c.cast( self.halsleep, c.c_void_p)
        icfg.cfg.atcacustom.halrelease = c.cast( self.halrelease, c.c_void_p)

        assert ATCA_SUCCESS == atcab_init(icfg)
        assert ATCA_SUCCESS == atcab_info(revision)
        print(''.join(['%02X ' % x for x in revision]))
        assert ATCA_SUCCESS == atcab_random(randomnum)
        print(''.join(['%02X ' % x for x in randomnum]))

    @c.CFUNCTYPE(c.c_int, c.c_void_p, c.c_void_p)
    def halinit(self, hal, cfg):
        pass

    @c.CFUNCTYPE(c.c_int, c.c_void_p) 
    def halpostinit(self, iface):
        pass

    @c.CFUNCTYPE(c.c_int, c.c_void_p, c.c_void_p, c.c_int)   
    def halsend(self, iface, txdata, txlength):
        pass

    @c.CFUNCTYPE(c.c_int, c.c_void_p, c.c_void_p, c.c_void_p,)  
    def halreceive(self, iface, rxdata, rxlength):
        pass
        
    @c.CFUNCTYPE(c.c_int, c.c_void_p) 
    def halwake(self, iface):
        pass 

    @c.CFUNCTYPE(c.c_int, c.c_void_p) 
    def halidle(self, iface):
        pass

    @c.CFUNCTYPE(c.c_int, c.c_void_p) 
    def halsleep(self, iface):
        pass
        
    @c.CFUNCTYPE(c.c_int, c.c_void_p) 
    def halrelease(self, haldata):
        pass

if __name__ == '__main__':
    ecc = Ecc()

```

bryan-hunt, 2019-02-14:
So I briefly looked at it (mostly to see if I could get pyserial to work with it) but I haven't spent much time on it. The biggest thing is you'll have to compile the library yourself since you'll need to have ATCA_HAL_CUSTOM defined when the library is built. 

So you can make the change to the lib/CMakeLists.txt file to add -DATCA_HAL_CUSTOM to cflags.

tjstone14, 2019-02-14:
Thanks, it looks like this might work.  I need to reformat the callbacks a bit, but now that I rebuilt per your instructions, I am able to see the library step through the callbacks.

tjstone14, 2019-02-27:
This ended up working well for me after rebuilding the library as suggested to include -DATCA_HAL_CUSTOM.  A nice-to-have would be to include this in the standard build so it is possible to to use the pip version of the library.

In case anyone is interested, here how I ended up structuring my HAL class.

`from cryptoauthlib import *
import ctypes as c

ATCA_SUCCESS = 0x00

class Ecc508:
    _icfg = ATCAIfaceCfg()

    def __init__(self):
        load_cryptoauthlib()

        Ecc508._icfg.devtype = ATCADeviceType.ATECC508A
        Ecc508._icfg.iface_type = 5
        Ecc508._icfg.cfg.atcacustom.halinit = c.cast( self.halinit, c.c_void_p)
        Ecc508._icfg.cfg.atcacustom.halpostinit = c.cast( self.halpostinit, c.c_void_p)
        Ecc508._icfg.cfg.atcacustom.halsend = c.cast( self.halsend, c.c_void_p)
        Ecc508._icfg.cfg.atcacustom.halreceive = c.cast( self.halreceive, c.c_void_p)
        Ecc508._icfg.cfg.atcacustom.halwake = c.cast( self.halwake, c.c_void_p)
        Ecc508._icfg.cfg.atcacustom.halidle = c.cast( self.halidle, c.c_void_p)
        Ecc508._icfg.cfg.atcacustom.halsleep = c.cast( self.halsleep, c.c_void_p)
        Ecc508._icfg.cfg.atcacustom.halrelease = c.cast( self.halrelease, c.c_void_p)

    @staticmethod
    @c.CFUNCTYPE(c.c_int, c.c_void_p, c.c_void_p,)
    def halinit(hal, cfg):
        return ATCA_SUCCESS

    @staticmethod
    @c.CFUNCTYPE(c.c_int, c.c_void_p) 
    def halpostinit(iface):
        return ATCA_SUCCESS

    @staticmethod
    @c.CFUNCTYPE(c.c_int, c.c_void_p, c.POINTER(c.c_ubyte), c.c_int)   
    def halsend(iface, txdata, txlength):
        return ATCA_SUCCESS

    @staticmethod
    @c.CFUNCTYPE(c.c_int, c.c_void_p, c.POINTER(c.c_ubyte), c.POINTER(c.c_ushort))  
    def halreceive(iface, rxdata, rxlength):
        return ATCA_SUCCESS

    @staticmethod
    @c.CFUNCTYPE(c.c_int, c.c_void_p) 
    def halwake(iface):
        return ATCA_SUCCESS 

    @staticmethod
    @c.CFUNCTYPE(c.c_int, c.c_void_p) 
    def halidle(iface):
        return ATCA_SUCCESS

    @staticmethod
    @c.CFUNCTYPE(c.c_int,c.c_void_p) 
    def halsleep(iface):
        return ATCA_SUCCESS

    @staticmethod
    @c.CFUNCTYPE(c.c_int, c.c_void_p) 
    def halrelease(haldata):
        return ATCA_SUCCESS`

bryan-hunt, 2019-02-27:
Thank you for reporting back on it. We’ll add the definition in so it’ll be available by default. Out of curiosity did you run any sort of memory profiler against it?

tjstone14, 2019-02-27:
I looked at it briefly using objgraph.  I didn't see any obvious issues, but this particular application is only open while it configures a device (about 10 sec) and then it closes.

closed 2019-02-27 by tjstone14