MicrochipTech/cryptoauthlib #55: atcab_verify_extern() verification failure

zacwbond, 2019-02-15:
I have had not had success using ```atcab_verify_extern()``` to verify a signature generated with openssl.  Here is my process:

First, I create a private key, extract the public key, and save it to a file:
```
openssl ecparam -name prime256v1 -genkey -out private.pem
openssl ec -in private.pem -pubout -outform PEM -out public.pem
```

Here is the public key in ```public.pem```:
```
-----BEGIN PUBLIC KEY-----
MFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAED/JwKRL+RTw8zpwRYqBJZoYBBTGj```
jJdOXcNH1GlDh83Pq1EeyEffhLZ3z/5pQ3se9ZQ8Gk0KrgXrUXbqsZfPgg==```
-----END PUBLIC KEY-----
```

My message is "hello."  Its SHA256 digest is as follows:
```
15 89 99 9B 0C A6 EF 88 14 28 30 26 A9 F1 66 D5
1C 70 A9 10 67 1C 3D 44 04 97 55 F0 7F 2E B9 10
```

If I save this to a binary file using HxD, I can sign it with my private key and verify it:
```
openssl dgst -sign private.pem -keyform PEM hash.bin > sig
openssl dgst -verify public.pem -signature sig hash.bin
```

I get a result of ```Verified OK```.  

Here is the contents of the file "sig" so that you can verify this yourself:
```
30 45 02 20 4A 09 A5 2E 71 6B 6F 87 FD 7F C5 C4 E2 73 
30 11 4E CA 52 94 CD F2 1E 55 96 5B 87 89 9B 74 49 2E
02 21 00 EB 6B B0 DF F8 2B 4D F4 9B 09 5E 5E 1B CC 21
0B 1B 91 E8 EA 3D 9B 72 6E 1E 49 12 48 EB 14 55 54
```

To use this with ```atcab_verify_extern()```, I need three C arrays.

The first is the SHA256 digest, which I generate using atcab_hw_sha2_256().  I verified that the results exactly match the contents of hash.bin above.

Next I need the signature.  From the file "sig" I can extract the r and s components into an array:

```
uint8_t signature[64] = {
    0x4A, 0x09, 0xA5, 0x2E, 0x71, 0x6B, 0x6F, 0x87, 0xFD, 0x7F, 0xC5, 0xC4,
    0xE2, 0x73, 0x30, 0x11, 0x4E, 0xCA, 0x52, 0x94, 0xCD, 0xF2, 0x1E, 0x55,
    0x96, 0x5B, 0x87, 0x89, 0x9B, 0x74, 0x49, 0x2E, 0xEB, 0x6B, 0xB0, 0xDF,
    0xF8, 0x2B, 0x4D, 0xF4, 0x9B, 0x09, 0x5E, 0x5E, 0x1B, 0xCC, 0x21, 0x0B,
    0x1B, 0x91, 0xE8, 0xEA, 0x3D, 0x9B, 0x72, 0x6E, 0x1E, 0x49, 0x12, 0x48,
    0xEB, 0x14, 0x55, 0x54
};
```

Finally, I need the public key's x and y components.  Using a B64 decode, I can extract those into the following array:

```
uint8_t publicKey[64] = {
	0x0F, 0xF2, 0x70, 0x29, 0x12, 0xFE, 0x45, 0x3C, 0x3C, 0xCE, 0x9C, 0x11,
	0x62, 0xA0, 0x49, 0x66, 0x86, 0x01, 0x05, 0x31, 0xA3, 0x8C, 0x97, 0x4E,
	0x5D, 0xC3, 0x47, 0xD4, 0x69, 0x43, 0x87, 0xCD, 0xCF, 0xAB, 0x51, 0x1E,
	0xC8, 0x47, 0xDF, 0x84, 0xB6, 0x77, 0xCF, 0xFE, 0x69, 0x43, 0x7B, 0x1E,
	0xF5, 0x94, 0x3C, 0x1A, 0x4D, 0x0A, 0xAE, 0x05, 0xEB, 0x51, 0x76, 0xEA,
	0xB1, 0x97, 0xCF, 0x82
};
```

Now I can call the function:
```
status =  atcab_verify_extern(hash, signature, publicKey, isVerified);
```
The result is a status of ```ATCA_SUCCESS```, but ```isVerified``` is 0 indicating that the signature is not verified.

What am I doing wrong?

BenUdall-Microchip, 2019-02-15:
Hi @zacwbond, it looks like you may be hashing your data twice. The openssl dgst command will also hash the data (defaults to md5 pre 1.1.0 and sha256 afterwards). Use your raw message "hello" as the input to openssl dgst and it should work.

BenUdall-Microchip, 2019-02-15:
I fixed up some notes I had on openssl and the ATECCx08A devices and posted them at https://github.com/MicrochipTech/cryptoauthlib/wiki/OpenSSL-and-ATECC-Sign-Verify

zacwbond, 2019-02-15:
That was indeed the problem.  Works great now.  Thanks for the quick response!

closed 2019-02-15 by zacwbond