MicrochipTech/cryptoauthlib #56: ATCA_PARSE_ERROR (0xD2) for ATECC608A AES-128 functions

unprovable, 2019-02-18:
Hello!

I'm running an ATECC608A off a raspberry pi (default installation of most recent, and updated raspbian) using I2C from the GPIO pins. 

The chip appears to be working, and I initialised it wit the stock config zone, but whenever I try to do any of the AES functions (including atcab_aes() ) from within python, I consistently get an ATCA_PARSE_ERROR (0xD2) returned.

Everything else appears to work - the RNG, SHA256 functionality, etc. Is there a way of troubleshooting this functionality?

Any help would be appreciated! 

Very best, M.

BenUdall-Microchip, 2019-02-18:
Hi @unprovable, ATCA_PARSE_ERROR (0xD2) can show up for a number of invalid parameters to the AES command (AES not enabled, invalid key location, etc...). Can you post your configuration zone and the parameters you're using for atcab_aes()?
```python
# Print config zone
config_data = bytearray(128)
assert ATCA_SUCCESS == atcab_read_config_zone(config_data)
print(' '.join(['{:02X}'.format(x) for x in config_data]))
```

unprovable, 2019-02-18:
Here's the output below:
```
>>> print(' '.join(['{:02X}'.format(x) for x in config_data]))
01 23 CB 25 00 00 60 02 5C 5E 1A D8 EE 01 19 00 C0 00 AA 00 81 A0 81 A0 81 A0 81 A0 81 A0 81 A0 81 A0 81 A0 81 A0 81 A0 81 A0 81 A0 81 A0 81 A0 81 A0 81 A0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 00 55 00 FF FF 00 00 00 00 00 00 33 00 33 00 33 00 33 00 33 00 33 00 33 00 33 00 3C 00 3C 00 3C 00 3C 00 3C 00 3C 00 3C 00 3C 00
```
The AES Enale bit is set (byte 13 is 0x01) but I think I see where the issue is - based on what you have said, all my keys are B283 NIST ECC type, I think? As such, I don't have any AES keys. I think I misunderstood the code here - I thought it would read a block from an ECC key as an AES key... I'm guessing you have to set the type explicitly! >.<

I don't suppose you can re-flash a config zone once it's locked, can you? If not, I have spare chips to move on to and test the AES with once I'm done with this one.

BenUdall-Microchip, 2019-02-18:
Ah! Thanks for the config. I see a number of things that could be issues:
- Your data zone isn't locked yet. The ATECC608A restricts many operations in this mode (config lock, data unlocked) as it's primarily meant for loading/generating initial keys and data. I suspect this is what is causing your parse error.
- The ATECC devices are also particularly cautious with ECC private keys and will not allow them to be used with any non-ECC command, such as AES. However, you only have slots 0-7 as ECC private keys. The slot you're trying to use, 10, is marked a symmetric SHA key (KeyType=0x7). Any slot to be used with the AES command needs to be set as AES (0x6) KeyType.

Small correction to your description, your private keys are setup as NIST P-256 (a.k.a. secp256r1 or prime256v1) ECC keys. This is the only curve the ATECC608A supports. B283 and K283 are only available in the ATECC108A.

The CryptoAuthentication devices (including the ATECC608A) can not be un-configured.

unprovable, 2019-02-18:
Ah, so this is indeed a problem with my misunderstanding rather than with the library or the chip! So to be clear (I'm really sorry about how silly I'm being):
- First set the config, and actually have some AES keys allocated, and lock the config zone
- Generate the keys into the slots, and then lock the data zone
- We should then be good to go? 
I'll close this issue now, as it's pretty much resolved :) Thank you so much for your help!!

closed 2019-02-18 by unprovable

BenUdall-Microchip, 2019-02-18:
Yup! Here's how I would describe the provisioning process:
- Write and lock the config zone. This sets the device and per-slot configuration/policies/permissions.
- Write and/or generate any initial keys and data. When the config zone is locked and data zone is unlocked, writes are allowed to all slots, but no reads and limited commands.
- Lock the data zone. This terminology may be a little misleading. It doesn't prevent changes to the slots and might be better described as enabling the policies and permissions of the slots set in the config zone. If the policies allow writes/changes to a slot then it can be updated with the appropriate command even after the data zone is locked.

unprovable, 2019-02-18:
Ah, that's cool! :D 
One last question if you don't mind me asking - is there an example for provisioning over Python? I used direct calls to libcryptoauth.so from a C prog last time, and as straightforward as that was (and I don't mind doing it) doing it through python would be, well, a bit more homogeneous (from a codebase point of view). 
In any case - many, many thanks for your help! I really like these ICs!

BenUdall-Microchip, 2019-02-18:
Certainly, all the standard calls in c should be available in python as well. The following example will give you a basic idea of provisioning in python: https://github.com/MicrochipTech/cryptoauthtools/blob/master/python/examples/config.py

Sleethers, 2019-11-03:
I know this is closed nearly a year now. But unprovable - is there a chance to get you config script ? I'm now at the very same problem but I can't create the keys ...
Sorry for bringing this up again

phanikarthik, 2020-07-03:
Does Secureboot and atcab_kdf work if datazone is not yet locked. (Config zone is locked).......In my case running these commands gives Execution_error