MicrochipTech/cryptoauthlib #57: Allow to build as either shared or static library

fredrec, 2019-02-18:
The type of library to build is currently hard-coded to `SHARED`:
`add_library(cryptoauth SHARED ${CRYPTOAUTH_SRC} ${ATCACERT_DEF_SRC})`

Some users want to build CryptoAuthLib as a static library instead.
A CMake option should allow that.

bryan-hunt, 2019-02-18:
Does the mbedtls branch work for you? That will be merged to mainline soon and that does have the option https://github.com/MicrochipTech/cryptoauthlib/blob/3bc5e245c68ca9f25e8fada393d059304b4adecb/lib/CMakeLists.txt#L65

fredrec, 2019-02-19:
> Does the mbedtls branch work for you? That will be merged to mainline soon and that does have the option

I did not notice that, yes it would work as well. 

A third option is to build both shared and static libraries. It can be done by first defining a library for the objects (`add_library(cryptoauth_objects OBJECT ${CRYPTOAUTH_SRC}`), then build both lib types:
```
add_library(cryptoauth SHARED $<TARGET_OBJECTS:cryptoauth_objects>)
add_library(cryptoauth_static STATIC $<TARGET_OBJECTS:cryptoauth_objects>)
```

Personally I have a preference for the `BUILD_SHARED_LIBS` way, but as long as it is possible to get a static library it is fine.

bryan-hunt, 2019-03-04:
This has been resolved by f86b9db with support for the option `ATCA_BUILD_SHARED_LIBS`.

closed 2019-03-04 by bryan-hunt