MicrochipTech/cryptoauthlib #60: atcab_sha fails at a size of over 4kb

habebit, 2019-02-20:
Is there a data size limitation for the atecc508a for hashing data? I've tested multiple data sizes and at a data size of about 4kb I get the `ATCA_GEN_FAIL= 0xE1, //!< unspecified error` error. I use the `atcab_sha` command.

Thank you very much
Habebit

BenUdall-Microchip, 2019-02-25:
Hi @habebit, I attempted to reproduce your error, but wasn't able to. I successfully hashed messages up to 1MB in size.

habebit, 2019-02-28:
@BenUdall-Microchip Thanks for your attemp. I use the atecc508a chip on a stm32 f767zi nucleo board via i2c connection. The strange part is that the atecc508a sometimes returns hashed messages up to 4096 Byte, but almost everytime it refuses it and I get the `0xE1` error. Maybe it's a limitation from the stm32 nucleo board. Thanks anyway.

closed 2022-12-03 by github-actions[bot]