MicrochipTech/cryptoauthlib #61: AWSIoTMQTTClient with PKCS#11

matriedel, 2019-02-24:
For the Greengrass-HSI you already providing a solution with PKCS#11 support

But I cannot have a full-blown data hungry Greengrass core on my IoT device as I only have a very restricted data rate mqtt connection with a small M2M data budget.

Having a look at AWS IoT Device SDK for Python you find one example for a client MQTT connection
https://aws.amazon.com/de/blogs/iot/how-to-implement-mqtt-with-tls-client-authentication-on-port-443-from-client-devices-python/

.# Import SDK packages
from AWSIoTPythonSDK.MQTTLib import AWSIoTMQTTClient

.# For certificate based connection
myMQTTClient = AWSIoTMQTTClient("myClientID")

.# Configure the MQTT Client
myMQTTClient.configureCredentials("YOUR/ROOT/CA/PATH", YOUR/DEVICE/KEY/PATH ", " YOUR/DEVICE/CERT/PATH ")

The API function call configureCredentials requests for a /DEVICE/KEY/PATH and does not offer any means to use PKCS#11 instead.

Instead of using MQTTClient.configureCredentials with a path pointing to a file that has the private key stored on the embedded system what can you recommend as an alternative way to configure the credentials by making use of PKCS#11?

Thank you for your advice

Best Matthias

bryan-hunt, 2019-02-26:
I don't currently have an answer for this one - it'll probably take some google searches to figure out if its at all possible to coerce the python ssl module to accept pkcs11 key formats and provide the write information to the underlying openssl library. It would probably require the global openssl configuration to be set up.

Native smartcard (pkcs11) support is on the list of things the python 3.8 team is looking at.

matriedel, 2019-02-27:
Is here eventually an alternative way based on OpenSSL engine support?
Something like instead of using the path you are pointing to the ENGINE...

privateKeyPath = "ENGINE"
AWSIoTMQTTClient.configureCredentials(rootCAPath, privateKeyPath, certificatePath)

Adding ENGINE support to the reconnect() method in client.py of AWSIoTPythonSDK
https://github.com/aws/aws-iot-device-sdk-python/blob/master/AWSIoTPythonSDK/core/protocol/paho/client.py

if self._tls_ca_certs is not None:
       if self._useSecuredWebsocket:
....
            else:
                  if self._tls_keyfile == "ENGINE":

=1=>  create SSL_CTX context object and load the certificates and keys into it
=2=>  Load private key via ENGINE_load_private_key
=3=>  Adding the private key to the context SSL_CTX_use_PrivateKey(context, privateKey)
=3=>  setup the connection OpenSSL.SSL.Connection(context, socket)
                      .
                      .
                      self._ssl.do_handshake()
....

Have you made some code available in that direction?

bryan-hunt, 2019-02-27:
These are questions for AWS or Eclipse (maintainers of Paho).

matriedel, 2019-02-28:
O.K thanks - I was asking as I thought your OpenSSL Engine implementation cryptoauth-openssl-engine https://github.com/MicrochipTech/cryptoauth-openssl-engine could be a fit for the ENGINE part

closed 2019-06-24 by bryan-hunt

zyfromsh, 2020-11-24:
Hi Matriedel, do you solve this? Currently, I am trying to use paho.mqtt.c to connect broker, device private key is stored in ATECC508 chip,while paho.mqtt.c uses openssl for security connection, I have no idea how paho.mqtt.c uses openssl to integrate with ATECC508

matriedel, 2020-11-24:
Hi zyfromsh
The discussion continued here https://github.com/awslabs/aws-c-io/issues/101 but unfortunately, a solution is still pending. Maybe you can ask there about a status update.