MicrochipTech/cryptoauthlib #62: emit warning when key is passed to JWT sign/encode

ptone, 2019-03-12:
as the key is properly ignored since signing is being done by the crypto element, might be worth emitting a python warning if anything other than "" is passed into the function so that people can catch improper calls.

bryan-hunt, 2019-03-12:
This is unfortunately not particularly cut and dry. HwEcAlgorithm is derived from ECAlgorithm which you can view here:

https://github.com/jpadilla/pyjwt/blob/ed28e495f937f50165a252fd5696a82942cd83a7/jwt/algorithms.py#L322-L373

The public key is required today not because the sign requires it but because the pyjwt module does in order to perform verification on the token:

https://github.com/jpadilla/pyjwt/blob/ed28e495f937f50165a252fd5696a82942cd83a7/jwt/api_jws.py#L111-L114

To support not providing the public key in the encode would have to be to replace the prepare_key and verify methods.

ptone, 2019-03-12:
ah yes - looking at https://github.com/jpadilla/pyjwt/blob/ed28e495f937f50165a252fd5696a82942cd83a7/jwt/algorithms.py#L334 this makes sense

closed 2022-12-03 by github-actions[bot]