MicrochipTech/cryptoauthlib #63: Additional HAL support.

RealityVoid, 2019-03-17:
Hello,

So I'm working with STM32 MCU's and doing some HAL for those for a project of mine. I guess you're not interested in adding HAL from competing HW vendors yourself, but would you be open to PR's from the community where we add additional HAL's?

habebit, 2019-03-17:
Can you provide you abstraction layer of your stm32 board? I'm currently working on a stm32 project with an atecc508a chip. But the wake command isn't working as expected.

schielimicrochip, 2019-03-17:
We typically provide implementations of abstraction layers for the Microchip Microcontrollers but not for those from our competitors. The example implementations should be of food help to “plug” i2c drivers from STM into the HAL

External E-Mail

RealityVoid, 2019-03-18:
@habebit I'm still working on it, and you might be further down the road with the HAL than I am. I just have some stupid things happening in my system right now and it's not working right yet. I was kind f hoping I'd find one somewhere, since there obviously are other people who need this.

@schielimicrochip I figured as much, but my question was - if I fork it and do the HAL in my fork would you be open on accepting PR's for these additional HALS? Or considering how they are your competitors you want no such things in your repo?

If I'm going there, I would like to know more about your license terms too, since it says this library should only be used with Microchip products. Do you consider using the atecc chips as sufficient to satisfy the terms of the license or would it be infringement to use this code on another vendor's MCU's?

RealityVoid, 2019-03-21:
@habebit so, I managed to take a look and I wake up the chip  properly usng the stm32 api. I now have write & read, wakeup and Idle, but it's still janky

my wakeup looks like this:

```
ATCA_STATUS hal_i2c_wake(ATCAIface iface)
{
    ATCAIfaceCfg *cfg = atgetifacecfg(iface);
    int retries = cfg->rx_retries;
    int status = HAL_ERROR;
    uint8_t data[4];

    // Send the 00 address as the wake pulse
    // part will NACK, so don't check for status
	HAL_I2C_Master_Transmit(&hi2c2, 0x00 ,data , 0, 100);

    // wait tWHI + tWLO which is configured based on device type and configuration structure
    atca_delay_ms(2);

    while (retries-- > 0 && status != HAL_OK)
    {
        status = HAL_I2C_Master_Receive(&hi2c2, cfg->atcai2c.slave_address, data, 4, 100);
    }

    if ((status != HAL_OK) && (status != HAL_TIMEOUT))
    {
        return ATCA_COMM_FAIL;
    }

    return hal_check_wake(data, 4);
	return ATCA_SUCCESS;
}
```

schielimicrochip, 2019-03-21:
We are not opposed to having such HAL in our repo. The main question is always how to support clients that would be using it. If it is pulled into our repo, we need to have evidence it went through proper testing.
Regarding the license, it is pretty straightforward: as soon as you are using this library with at least the CryptoAuthentication device, you meet the licensing requirements. It does not have to be the MCU.

External E-Mail

SameerChilmattur, 2019-10-24:
> @habebit so, I managed to take a look and I wake up the chip properly usng the stm32 api. I now have write & read, wakeup and Idle, but it's still janky
>
> my wakeup looks like this:
>
> ```
> ATCA_STATUS [...]

having trouble with flaky i2c from stm32 to 508a. Does the 508a have timing constraints on wake cmd? 
on a side note, those 2 return statements are interesting :)

RealityVoid, 2019-10-31:
I did the thing I had to do with cryptauthlib on STM32 quite a while ago so my memory is hazy. I do remember that indeed, there were some timing constraints related to the wake up, but that depends to the configuration you're loading. 

I did manage to get the HAL to fully comply to what I wanted to do. And I added some other changes. Butthen it just lost interest in it so I omitted to put it on GitHub. I'll see later this evening if I can still find the latest thing I used and maybe I'll put my branch here, who knows maybe some people will find it useful somehow. 

Heh, yes, the 2x return statement was just me being stupid.

bullitmaster, 2019-12-26:
Hi!

I would like to know if there are plans to add third-party HAL support?
I have a gpio and uart stm32 driver using stm32 HAL. Supporting (F4 and F7 series). Now I am testing them, and soon I am ready to push them out. BUT at the moment, without making changes to the library, compiling them is difficult!
Can you give me advice on how i can builsd it now without making changes?

Best Regards,
bullitmaster!

bullitmaster, 2019-12-26:
Supporting F4xx, F7xx and H7 series.

SenZCS, 2022-11-11:
Hi, I am trying to interface ATECC608B with STM32 using cryptoauthlib. How do I map the I2C of STM32 in the cryptoauthlib?

bkkrunal, 2023-01-11:
Hi @RealityVoid 
I have successfully compiled the library with my stm32 HW and now I have been trying very hard to wakeup ATECC608 using this library. However it just doesn't seem to be waking up and it doesn't respond with wakeup data.
Any chance you have got this working on your HW.
I would highly appreciate your help.

Thanks.

closed 2023-03-23 by github-actions[bot]

dnssoftware, 2023-10-19:
> @habebit so, I managed to take a look and I wake up the chip properly usng the stm32 api. I now have write & read, wakeup and Idle, but it's still janky
>
> my wakeup looks like this:
>
> ```
> ATCA_STATUS [...]

I have tried this and I don't understand why, it doesn't wake up my ATECC608B.
Has anyone got a full sample project or other pointers?

Using STM32G4 chip.

The same hardware is being woken up by an arduino build, which is not what I'd like to use.

RealityVoid, 2023-10-19:
@dnssoftware  it's been an _eternity_ since I saw one of those chips, so take this with a grain of salt, but I might have a couple of hints: 

- Check your tWHI  configurations, depending on that you might want to wait a bit longer to check the response from slave. 
- I noticed I stupidly did not init data array, try doing that, I don't remember if the wakeup request was just 00 address or the data after was needed as well to be zero. 
- if you're breaking somewhere with a jtag debugger, do be careful, it might throw you back into sleep before you resume and you might _think_ it did not work, when it actually did. 
- Worst case, put a logic analyser on the I2C bus and see what's going on there. 

Sorry I can't help more.

Best,
Vlad R.

rikilshah, 2023-11-10:
> I have tried this and I don't understand why, it doesn't wake up my ATECC608B. Has anyone got a full sample project or other pointers?
>
> Using STM32G4 chip.
>
> The same hardware is being woken up by an [...]

I am currently working on this Chip and initially, I also faced the issue of not getting any response. What helped me was to set I2C bus frequency to 100khz. 

If anyone has done any work on STM32+ATECC608a, I would be interested to know further details.

MAD-Riccardo, 2024-03-18:
Io ho lavorato e sto lavorando con stm32F7 e H7 e con ATECC508A e ATECC608A.
Se qualcuno avesse bisogno di supporto mi scriva.

FaustoYair, 2024-07-29:
> Io ho lavorato e sto lavorando con stm32F7 e H7 e con ATECC508A e ATECC608A. Se qualcuno avesse bisogno di supporto mi scriva.

Hi MAD-Ricaardo, how can I contact you? I'm also working but I've had problems using the library.

MAD-Riccardo, 2024-07-29:
g.riccardo@madengineeringsrl.it