MicrochipTech/cryptoauthlib #65: atcab_genkey return ATCA_EXECUTION_ERROR

Cooper-WJH, 2019-03-20:
I use espressif ESP32 to control ATECC608A.

The `atcab_random` & `atcab_read_serial_number` work fine, but when I use `atcab_genkey` it return `ATCA_EXECUTION_ERROR`. What's the problem is?

```c
int32_t atecc608a_ecdh_genkey(uint8_t *pubKey)
{	
	ATCA_STATUS status;
	uint8_t pub_alice[ATCA_PUB_KEY_SIZE];
	uint8_t frag[4] = { 0x44, 0x44, 0x44, 0x44 };

	memset(pub_alice, 0x44, ATCA_PUB_KEY_SIZE);
	status = atcab_genkey(KEY_ID, pubKey);
	MDF_ERROR_CHECK(status != ATCA_SUCCESS, MDF_FAIL, "atcab_genkey fail: %d", status);

	MDF_LOGI("=== ATECC608A generate key ===");
	ESP_LOG_BUFFER_HEX(TAG, pub_alice, ATCA_PUB_KEY_SIZE);

	return MDF_OK;
}
```

bryan-hunt, 2019-03-21:
So it's clear that communication is working. If you retry the command you get the same failure the execution error is indicative of a configuration/usage problem. To get hardware/product support please through your assigned FAE or submitted as a request through the mySupport portal on [microchip.com](http://www.microchip.com/support/hottopics.aspx). 

For details on the mySupport Portal or how to use it please see the [instructions for using the mySupport portal](https://microchip.secure.force.com/microchipknowledge/articles/en_US/FAQ/How-to-submit-a-case).

JacobBai, 2020-11-11:
Hi Cooper,

Did you solve the problem?

I got the same error with raspberry pi.

Jacob

closed 2020-12-03 by bryan-hunt

Hikari1026, 2021-06-08:
Hello everyone, I'm sorry to reopen this, but I'm encountering a similar issue on Arduino Uno. I'm able to execute flawlessly commands like `atcab_random`, `atcab_read` and other "basic" ones as well, but whenever I try to execute a GenKey in order to calculate slot's public key, I get some weird errors:

- If an ATCECC608A is connected to the Arduino, it returns error code `ATCA_WAKE_SUCCESS`
- If an ATECC608B-TFLXTLS is connected instead, the script stalls forever like it's waiting to receive the data from the device

I'm assuming both configuration and device are working properly, since I'm able to execute the GenKey and Sign commands with a SAM D21 Xplained pro kit, so I'm guessing it's a communication issue with this command via I2C.

Did you managed to find a fix? Thank you in advance for your time and help.