MicrochipTech/cryptoauthlib #66: Python 3.7.2 OpenWrt cryptoauthlib/libcryptoauth.so: Exec format error

achandel, 2019-03-20:
After installing package on following:
CPU: Mediatek 7688 Arch: MIPS R32
OS: OpenWrt 18.6

Error in  from cryptoauthlib import *

**"Exec format error"**

Traceback (most recent call last):
    from cryptoauthlib import *
  File "/site-packages/cryptoauthlib/__init__.py", line 16, in <module>
    load_cryptoauthlib()
  File "/site-packages/cryptoauthlib/library.py", line 82, in load_cryptoauthlib
    _CRYPTO_LIB = ctypes.cdll.LoadLibrary(os.path.join(curr_path, "libcryptoauth.so"))
OSError: Error loading shared library/site-packages/cryptoauthlib/libcryptoauth.so: Exec format error

bryan-hunt, 2019-03-20:
So my first guess is that something went wrong during the build. Did you cross compile or build on target?

You can use readelf to dump the basic information for which arch the library is built. Also you can use ldd to see if any of the expected dependencies are missing on the target.

If it was built on target I would expect it to end up a different arch (but it’s worth checking to make sure cmake detected your platform properly) and the link step would have definitely failed without the dependencies.

achandel, 2019-03-20:
We cross compiled. I I'll try compiling it on target and let you know.

closed 2019-06-24 by bryan-hunt