MicrochipTech/cryptoauthlib #67: Secret-dependent comparisons should be constant-time

CarlMitchellKT, 2019-03-21:
memcmp() is used in several areas where a constant-time equivalent would be safer.

bryan-hunt, 2019-03-21:
Can you be more specific for which areas in code that are presenting a concern?

CarlMitchellKT, 2019-03-21:
I made the issue to have some context for the pull request. Particularly atcab_verify(), since it's dealing with ECDSA signatures directly.

bryan-hunt, 2019-03-21:
We can provided it as an option for the particularly paranoid users over the platform specific version they may currently be using. As its impractical to create a sha256 rainbow table and if an attacker has physical access with high precision timing - it will be simply easier for them to read the secrets right out the MCU ram or flash especially since the recommendation is to use a random nonce with a mac making replays impossible.

bryan-hunt, 2019-03-21:
Also after reviewing the changes each of these comes after a operation that is controlled by the timing of the device which is protected against timing and DPA attacks already. At that point the memcmp timing is irrelevant. An attacker would have to instead intercept the result and then perform their own brute force against the mac.

bryan-hunt, 2019-03-21:
An important clarification - there is nothing secret in an ECDSA verification - both the signature and the public key are by definition public and anybody can perform a verification. The ATECC608A has an optional mode that will provide a MAC in response with the verification to eliminate a man in the middle providing a direct verification response that did not come from the device.

CarlMitchellKT, 2019-03-21:
Good points. I'll close this issue. Thanks for the prompt assistance and information (some of which I know but clearly needed more caffeine to remember.)

closed 2019-03-21 by CarlMitchellKT