MicrochipTech/cryptoauthlib #69: AES-CCM support for ATECC608A

DurandA, 2019-03-22:
The [ATECC608A product description](https://www.microchip.com/wwwproducts/en/ATECC608A) mentions AES-CCM. However, I don't see any `atcab_aes_ccm*` function. Do you plan to support AES-CCM in *CryptoAuthLib*?

BenUdall-Microchip, 2019-03-22:
Hi @DurandA , I think that's actually a typo on the page. Should probably say AES-GCM. Either way, the device does support the AES128 ECB and CryptoAuthLib implements a number of block cipher modes around that. Currently we support CBC, CTR, and GCM. CCM is in the works, but not ready yet.

DurandA, 2019-03-22:
Do you mean that ATECC608A has hardware support for AES-CCM and that you will add software support for it? Or will Microchip release a new device?

BenUdall-Microchip, 2019-03-22:
ATECC608A has hardware support for AES-ECB, which will be used by a software implementation of CCM in CryptoAuthLib.  Just requires a software/CryptoAuthLib update.

DurandA, 2019-03-25:
Awesome! Do you have any ETA of AES-CCM support?

DurandA, 2019-08-02:
Is this still on track?

DurandA, 2019-11-08:
@BenUdall-Microchip @bryan-hunt I just saw that Microchip released ATECC608A-TNGLORA. LoRaWAN requires AES-CCM and AES-CMAC, is there any LoRaWAN stack available using CryptoAuthLib?

schielimicrochip, 2019-11-08:
Yes, Microchip has a loRaWAN stack running on the SAMR34 for instance that is using cryptoauthlib. Similarly, ARM has integrated it into their LoRAWAN stack that is part of the mbed stack

Regards
Nicolas Schieli
438 505-3591

External E-Mail

DurandA, 2019-11-08:
@schielimicrochip Could you upstream the AES-CCM (and CMAC) implementation from the SAMR34 LoRaWAN stack to CryptoAuthLib?

bryan-hunt, 2021-06-21:
CCM and CMAC support were added quite a while ago (v3.2.5) but this issue had never been updated.

closed 2021-06-21 by bryan-hunt