MicrochipTech/cryptoauthlib #70: Support for setting things like i2s bus and address from a config file

jonsmirl, 2019-04-06:
We're looking at adding cryptoauthlib into buildroot as a build option, but that's incompatible with having to edit the source code to configure the device address (like i2c bus/address).  Buildroot is a Linux userspace environment.  https://buildroot.org/

Buildroot recently added libp11 p11-kit support.

BenUdall-Microchip, 2019-04-08:
Hi @jonsmirl, the I2C address (and other device connection info) was intended to be set from the application. The built-in [ATCAIfaceCfg](https://github.com/MicrochipTech/cryptoauthlib/blob/f86b9dbb1e805d85a435456ca03450342dbdc43e/lib/atca_iface.h#L66) structures in [atca_cfgs.c](https://github.com/MicrochipTech/cryptoauthlib/blob/master/lib/atca_cfgs.c) were meant to be examples or defaults, but are not required.

It's expected an application would do something like the following instead of modifying the source code:
```c
ATCAIfaceCfg config = cfg_ateccx08a_i2c_default;  // Init config with default I2C params
config.atcai2c.slave_address = 0x6E;              // Set I2C address
atca_init(&config);                               // Init interface
```

jonsmirl, 2019-04-10:
How about the bus number? It defaults to bus 1, I had to edit the source to switch it to bus 0.

jonsmirl, 2019-04-10:
diff --git a/lib/atca_cfgs.c b/lib/atca_cfgs.c
index 79cf286..2df62eb 100644
--- a/lib/atca_cfgs.c
+++ b/lib/atca_cfgs.c
@@ -42,7 +42,7 @@ ATCAIfaceCfg cfg_ateccx08a_i2c_default = {
     .iface_type             = ATCA_I2C_IFACE,
     .devtype                = ATECC608A,
     .atcai2c.slave_address  = 0xC0,
-    .atcai2c.bus            = 1,
+    .atcai2c.bus            = 0,
     .atcai2c.baud           = 400000,
     //.atcai2c.baud = 100000,
     .wake_delay             = 1500,

BenUdall-Microchip, 2019-04-10:
All those values can be set in the application.
```c
ATCAIfaceCfg config = cfg_ateccx08a_i2c_default;  // Init config with default I2C params
config.atcai2c.slave_address = 0x6E;              // Set I2C address
config.atcai2c.bus           = 0;                 // Set bus
atca_init(&config);                               // Init interface
```

jonsmirl, 2019-04-10:
How do I set the bus when use the 0.conf file in the pk11 support?

interface = i2c,0xC0
freeslots = 1,2,3

bryan-hunt, 2019-04-10:
This is an open item that will be addressed in the next release. Depending on your use case a static configuration may be more useful (I.e you’re using yocto or buildroot) where the filesystem configuration is instead defined at compile time. A wiki article for this is forthcoming that will describe the data model for static configuration.

jonsmirl, 2019-04-10:
I am patching the source code in my buildroot environment.  The buildroot people want to add crytoauthlib as an option in buildroot. In that case patching the source is not a general purpose solution.

bryan-hunt, 2020-12-03:
This should have been resolved quite a while ago - however it is more clearly documented in the template now

closed 2020-12-03 by bryan-hunt