MicrochipTech/cryptoauthlib #78: secure_boot example broken at tag

DavidZemon, 2019-04-30:
There are header files being included that do not exist (and are not needed). The following patch fixes it:

```.patch
Index: app/secure_boot/secure_boot.h
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
--- app/secure_boot/secure_boot.h	(date 1551726444000)
+++ app/secure_boot/secure_boot.h	(date 1556660267000)
@@ -36,7 +36,6 @@
 #include "atca_status.h"
 #include "secure_boot_memory.h"
 #include "atca_command.h"
-#include "crypto\atca_crypto_sw_sha2.h"
 #include "host/atca_host.h"
 
 
Index: app/secure_boot/secure_boot.c
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
--- app/secure_boot/secure_boot.c	(date 1551726444000)
+++ app/secure_boot/secure_boot.c	(date 1556660269000)
@@ -31,9 +31,7 @@
 #include "secure_boot.h"
 #include "atca_iface.h"
 #include "hal/atca_hal.h"
-#include "test/atca_test.h"
 #include "io_protection_key.h"
-#include "crypto_device_app.h"
 
 /*Initialization routines */
 static ATCA_STATUS secure_boot_init(secure_boot_parameters* secure_boot_params);
```

In `secure_boot.h`, the file does exist but the backslash causes a failure in Linux. The extra headers in `secure_boot.c` either don't exist or are not being included by my custom CMake script for building the cryptoauth library. Maybe I could fix it by adding more include directories to the library target, but since the headers don't seem to be required anyway, I thought it was better to just remove them.

KalyanCManukonda, 2019-05-02:
You are right backslash is an issue in Linux. This needs to be fixed. Also, there are extra headers exist that we don't need. After revisiting the implementation, Following changes would be more appropriate. Let me know if these works for you

```.patch
--- a/app/secure_boot/secure_boot.c
+++ b/app/secure_boot/secure_boot.c
@@ -26,14 +26,10 @@
  * THIS SOFTWARE.
  */
 
-#include <stdlib.h>
-#include <stdio.h>
+#include <string.h>
 #include "secure_boot.h"
-#include "atca_iface.h"
-#include "hal/atca_hal.h"
-#include "test/atca_test.h"
 #include "io_protection_key.h"
-#include "crypto_device_app.h"
+#include "basic/atca_basic.h"
 
 /*Initialization routines */
 static ATCA_STATUS secure_boot_init(secure_boot_parameters* secure_boot_params);
=========================================================== 
 
--- a/app/secure_boot/secure_boot.h
+++ b/app/secure_boot/secure_boot.h
@@ -36,9 +36,7 @@ extern "C" {
 #include "atca_status.h"
 #include "secure_boot_memory.h"
 #include "atca_command.h"
-#include "crypto\atca_crypto_sw_sha2.h"
-#include "host/atca_host.h"
-
+#include "crypto/atca_crypto_sw_sha2.h"
 
 #define SECURE_BOOT_CONFIG_DISABLE              0
 #define SECURE_BOOT_CONFIG_FULL_BOTH            1
```

DavidZemon, 2019-05-02:
Sure, seems. Thanks

closed 2022-12-02 by github-actions[bot]