MicrochipTech/cryptoauthlib #79: Example implementations and/or documentation for functions in secure_boot_memory.h

DavidZemon, 2019-04-30:
There seems to be no documentation as to what the functions in `secure_boot_memory.h` are supposed to do, and I'm unable to find any example implementations of these functions to reference.

I'd sure appreciate any help you can give me on figuring out how to implement these functions for my app.

bryan-hunt, 2019-04-30:
You can find more details for use in https://github.com/MicrochipTech/cryptoauth_usecase_secureboot

DavidZemon, 2019-04-30:
Okay, that got me to https://github.com/MicrochipTech/cryptoauth_usecase_secureboot/blob/master/SAMBA_BOOTLOADER/SAMBA_D21_BOOTLOADER1/src/secure_boot_memory.c, which has an example implementation of each function. That's a start, and I suppose if I stare at it long enough and start researching all the function calls that are specific to this project it might start making sense.

bryan-hunt, 2019-04-30:
For the most part the explanation for how to utilize those functions are part of the overall wiki for that usecase: https://github.com/MicrochipTech/cryptoauth_usecase_secureboot/wiki

DavidZemon, 2019-04-30:
That looks like the same content as the application note, [8984](http://ww1.microchip.com/downloads/en/AppNotes/Atmel-8984-CryptoAuth-CryptoAuthLib-ApplicationNote.pdf), and the only related bit is on https://github.com/MicrochipTech/cryptoauth_usecase_secureboot/wiki/Porting-to-other-Bootloaders, which simply says

> 4. Update memory interfaces:
>    * Secure_boot_memory.c and io_protection_key.c files contain access functions for the NVM/Flash. These access functions should be revisited based on the device in use.

Am I missing something else in there that's more descriptive?

bryan-hunt, 2019-04-30:
The challenge here is that flash access is highly implementation specific of the hardware. If you're using a Microchip MCU you can always submit a support request through myMicrochip support: https://www.microchip.com/mymicrochip/

@KalyanCManukonda Do you have any additional documentation for what the required tasks and outputs are expected from the MCU porting layer?

DavidZemon, 2019-04-30:
An example that just verifies a small block of memory in RAM would be fantastic. That would be enough to explain what the functions are supposed to do, and then I'll (hopefully) know enough to port it to my own application.

KalyanCManukonda, 2019-05-02:
Other than this wiki, there is no extra documentation available. 

https://github.com/MicrochipTech/cryptoauth_usecase_secureboot/wiki/Building-and-Downloading-the-User-Application page provides steps to compile User application with required changes for this bootloader. 

Coming to secure_boot_memory.c contains apis that are device and application specific like flash/nvm access apis, enabling boot protections after bonding with secure element, get application memory parameters (memory_parameters) like its start address, length etc. io_protection_key.c contains IO protection key access apis. In the example, we simply write/read flash.

closed 2022-12-02 by github-actions[bot]