MicrochipTech/cryptoauthlib #82: writing a public key via pkcs11 fails

raerne, 2019-05-15:
When attempting to write a public key via `p11tool` i noticed the following problem. In `pkcs11_key_write` it is assumed that the key data is embedded into an ASN.1 SubjectPublicKeyInfo type defined as follows [[RFC5480]](https://tools.ietf.org/html/rfc5480#section-2)
```
SubjectPublicKeyInfo  ::=  SEQUENCE  {
       algorithm         AlgorithmIdentifier,
       subjectPublicKey  BIT STRING
     }
```

which is encoded as follows (if a 64byte key is used)
```
/** ASN.1 Header for SECP256R1 public keys */
static const uint8_t ec_pubkey_asn1_header[] = {
0x30, 0x59, 0x30, 0x13, 0x06, 0x07, 0x2A, 0x86,
0x48, 0xCE, 0x3D, 0x02, 0x01, 0x06, 0x08, 0x2A,
0x86, 0x48, 0xCE, 0x3D, 0x03, 0x01, 0x07, 0x03,
0x42, 0x00, 0x04};
```

however, `p11tool` simply provides the OCTET-STRING which has the much shorter header of
```
static const uint8_t ec_pubkey_asn1_octet_string_header[] = {
0x04, 0x41, 0x04
};  
```

The following patch fixes the problem of writing the key, however I am unsure how the PKCS11 specification defines this, i.e. is it a problem with `p11tool` or this library

```
---
 lib/pkcs11/pkcs11_key.c | 8 ++++++--
 1 file changed, 6 insertions(+), 2 deletions(-)

diff --git a/lib/pkcs11/pkcs11_key.c b/lib/pkcs11/pkcs11_key.c
index 45b418f..9378177 100644
--- a/lib/pkcs11/pkcs11_key.c
+++ b/lib/pkcs11/pkcs11_key.c
@@ -149,6 +149,10 @@ static const uint8_t ec_pubkey_asn1_header[] = {
     0x42, 0x00, 0x04
 };
 
+static const uint8_t ec_pubkey_asn1_octet_string_header[] = {
+    0x04, 0x41, 0x04
+};
+
 /**
  * \brief Extract a public key and convert it to the asn.1 format
  */
@@ -545,9 +549,9 @@ CK_RV pkcs11_key_write(CK_VOID_PTR pObject, CK_ATTRIBUTE_PTR pAttribute)
 
     if (obj_ptr->class_id == CKO_PUBLIC_KEY)
     {
-        if(!memcmp(ec_pubkey_asn1_header, pAttribute->pValue, sizeof(ec_pubkey_asn1_header)))
+        if(!memcmp(ec_pubkey_asn1_octet_string_header, pAttribute->pValue, sizeof(ec_pubkey_asn1_octet_string_header)))
         {
-            status = atcab_write_pubkey(obj_ptr->slot, &(((uint8_t*)pAttribute->pValue)[sizeof(ec_pubkey_asn1_header)]));
+            status = atcab_write_pubkey(obj_ptr->slot, &(((uint8_t*)pAttribute->pValue)[sizeof(ec_pubkey_asn1_octet_string_header)]));
         }
         else
         {
--
2.7.4 
```
Notice that to test this, PR #81 needs to be fixed also. Then a key can be written, e.g. to slot 15 when using the default config (if it is configured as a freeslot, with label root and no slotconfig exists, i.e. no file /var/lib/cryptoauthlib/0.15.config exists)
```
p11tool --load-pubkey pubkey.pem --write "pkcs11:token=..." --label root
```

bryan-hunt, 2019-05-16:
This is an odd one and I have confirmed the behavior but I have not confirmed which is actually correct in the specification. If the EC point into the api is different than the EC point out of the API that would be really confusing.

bryan-hunt, 2019-05-16:
Okay I have confirmed (pkcs11-curr-v2.40-errata01-os-complete section 2.3.3) that the bug is only in pkcs11_key_write and the remainder of the implementation uses the DER encoding of ANSI X9.62 ECPoint value Q.

bryan-hunt, 2019-05-24:
This should be resolved in the updated pkcs11 branch in cf58f31

commit cf58f31b (2019-05-24): Merge PKCS11 patches from other projects
* Return the actual certificate size rather than the maximum possible size
* Finalized static configuration API and moved example to app/pkcs11 folder
* Fix public key write (resolves #82)
* Fix freeslot parsing (resolves #81)
* Extend i2c interface config to include bus and speed (#70)
closed 2019-06-24 by bryan-hunt