MicrochipTech/cryptoauthlib #83: Will the branch PKCS11 be merged to master or updated?

Rasnar, 2019-05-16:
Hi,

I'm currently working on a project using ATECC508A and using QNX as the OS.

The master branch is fine, with very small modifications I can compile it correctly (I'm preparing a pull request), thanks to the multiples improvements proposed in some of the issues about hid disabled options. However, the branch PKCS11 is way behind in commits compared to master and don't have all those improvements. 

Is there any plan to re-sync them together soon or will they keep living separately? Since the cryptoauth-openssl-engine is considered deprecated and that we should use PKCS11 option, I'm left with this option for now.
If there is no plan in the near future I will try to merge master into PCKS11 to get all the improvements for PCKS11.

bryan-hunt, 2019-05-16:
There is a pending update to cryptoauthlib that will be merged into the pkcs11 branch as it contains some needed updates. I'll update this issue when those changes have been made.

bryan-hunt, 2019-06-04:
@Rasnar The pkcs11 branch has been updated with the latest release of cryptoauthlib.

SuicidalLabRat, 2019-09-05:
Bump.  
Given the recent release to master, is there any update on whether this will merge into master, or visa versa.

ambarus, 2020-02-20:
Other problem with having this branch split from master is that if not merging it, you'll need some tags with release versions on pkcs11 too. Is the latest commit always guaranteed to work?

nashneshev, 2020-12-09:
@bryan-hunt You have updated [PKCS11 Linux Setup](https://github.com/MicrochipTech/cryptoauthlib/wiki/PKCS11-Linux-Setup) instructions to use the main branch when setting up cryptoauthlib as a pkcs11 provider. So what is the purpose of the pkcs11 branch now? Should we be concerned with it as cryptoauthlib users?
I am using the library both as a pkcs11 provider and to directly talk to the devices and up until now I had two separate versions for that.

bryan-hunt, 2020-12-09:
@nashneshev it exists simply because deleting a branch or commit someone might expect is generally considered bad - As we are nearing a year of V3.2.x the time remaining for the pkcs11 branch is becoming very short indeed. PKCS11 support is available in the mainline now (which is versioned and tagged) so the instructions were updated to use what is available in main for new users of the library. You can safely update your systems to only use one copy of the library.

closed 2022-12-02 by github-actions[bot]