MicrochipTech/cryptoauthlib #85: pkcs11 : RNG dies not work

mickaelchazaux, 2019-06-04:
pkcs11-tool --module /lib/libcryptoauth.so --generate-random 1000 | hexdump

Expected : Random bytes

Got : 

0000000 ffff 0000 ffff 0000 ffff 0000 ffff 0000
*
0000020 0000 0000 0000 0000 0000 0000 0000 0000
*
00000e0 0000 0000 9a08 b6c7 0170 0000 b8c1 0001
00000f0 0023 6552 6573 7672 6465 4300 6e6f 6966
0000100 7567 6172 6974 6e6f 6600 726f 6100 6400
0000110 7665 6369 0065 0023 6854 0065 626f 656a
0000120 7463 0073 6e69 7400 6968 0073 6966 656c
0000130 7700 6c69 006c 6562 6300 6572 7461 6465
0000140 6100 646e 6d00 7261 656b 0064 7361 7500
0000150 646e 6c65 7465 6261 656c 2300 5400 6568
0000160 6573 6100 6572 7000 6f72 6563 7373 6465
0000170 6900 006e 726f 6564 2e72 4300 6e6f 6966
0000180 7567 6172 6974 6e6f 7000 7261 6d61 7465
0000190 7265 0073 756d 7473 6200 0065 6f63 6d6d
00001a0 0061 0023 6564 696c 696d 6574 0064 6e61
00001b0 0064 616d 0079 6f6e 0074 6f63 746e 6961
00001c0 006e 7073 6361 7365 0000 6e69 6574 6672
00001d0 6361 0065 0000 3269 0063 7830 3043 3000
00001e0 3100 3030 3030 0030 7266 6565 6c73 746f
00001f0 0073 0000 0031 0032 0033 2300 5300 6f6c
0000200 0074 0030 7369 7400 6568 7000 6972 616d
0000210 7972 7000 6972 6176 6574 6b00 7965 6f00
0000220 6a62 6365 0074 0000 7270 7669 7461 0065
0000230 6564 6976 6563 3000 0000 0023 6c53 746f
0000240 3100 0030 7369 7400 6568 6300 7265 6974
0000250 6966 6163 6574 6400 7461 0061 6f66 0072
0000260 6874 0065 6564 6976 6563 7327 7000 6275
0000270 696c 0063 656b 0079 6f23 6a62 6365 0074
0000280 0000 6563 7472 6669 6369 7461 2c65 6564
0000290 6976 6563 312c 0030 2300 5300 6f6c 0074
00002a0 3231 6900 0073 6874 0065 6e69 6574 6d72
00002b0 6465 7461 2f65 6973 6e67 7265 6300 7265
00002c0 6974 6966 6163 6574 6400 7461 0061 6f23
00002d0 6a62 6365 0074 0000 6563 7472 6669 6369
00002e0 7461 2c65 6973 6e67 7265 312c 0032 2300
00002f0 5300 6f6c 0074 3531 6900 0073 0061 7570
0000300 6c62 6369 6b00 7965 6f00 6a62 6365 0074
0000310 0000 7570 6c62 6369 7200 6f6f 0074 3531
0000320 0000 0000 b689 0001 0000 0000 0000 0000
0000330 0000 0000 0000 0000 0000 0000 0000 0000
*
00003e0 0000 0000 0000 0000                    
00003e8

In ascii : 

pkcs11-tool --module /lib/libcryptoauth.so --generate-random 1000          
Using slot 0 with a present token (0x0)
���������������*Ķp�#ReservedConfigurationforadevice#Theobjectsinthisfilewillbecreatedandmarkedasundeletable#Theseareprocessedinorder.Configurationparametersmustbecomma#delimitedandmaynotcontainspacesinterfacei2c0xC00100000freeslots123#Slot0istheprimaryprivatekeyobjectprivatedevice0#Slot10isthecertificatedataforthedevice'spublickey#objectcertificate,device,10#Slot12istheintermedate/signercertificatedata#objectcertificate,signer,12#Slot15

bryan-hunt, 2019-06-04:
The first bytes are the pattern the device returns when you haven’t configured it. See section 3.3.2 of the [datasheet](http://ww1.microchip.com/downloads/en/DeviceDoc/20005927A.pdf).

As the complete buffer should have been filled and clearly was not I will investigate what in the update changed this behavior

bryan-hunt, 2019-06-04:
@mickaelchazaux With the update you should expect to see a repeating pattern of ffff 0000 through the entire buffer until the device has been configured.

mickaelchazaux, 2019-06-11:
Yes, it works. 

Thank you

PR #87 (2019-06-04): Restore windows build and token_random functionality
Windows PKCS11 isn't supported but the CMakeLists.txt should be correct for the python build when it is merged back.

Resolves #85 as well.
closed 2019-06-11 by mickaelchazaux