MicrochipTech/cryptoauthlib #86: PKCS11 : What are the supported features ?

mickaelchazaux, 2019-06-04:
I am evaluating the ATECC508 device for inclusion in our product line and I find that the PKCS#11 support is quite lacking.

What are the features of the PKCS#11 implementation for this device ? 
Specifically : 
 - Is initializing the token with a SO-PIN supported ?
 - Is resetting/changing the SO-PIN supported ? 
 - Is Setting/Resetting/Changing the USER PIN supported ?
Once initialized : 
 - Is generating new keys supported ? 
 - Is deleting keys supported ?
 - Is importing/deleting certificates supported ?
 - Does the PIN must be presented to the device to use the keys ?
 - Is it possible to re-initialize (recycle) the device, erasing all keys and setting a new SO-PIN ?

From what I read from the code & doc, the ATECC508 must be "factory programmed" with keys that will never be able to be changed. Is that correct ? Does this mean "No" to all questions above ?

Best regards, 

Mickael

schielimicrochip, 2019-06-04:
I would suggest you contact your local Microchip sales office to get access to the ECC608A datasheet. It will allow you to get most of the device related questions answered.
The ECCx08A does not “have” to be factory programmed. You can elect to use Microchip’s secure provisioning flow but you are entirely free to do it yourself.

Regards
Nicolas

bryan-hunt, 2019-06-04:
First one has to have a good understanding of the ECC devices from Microchip. The ATECC508A & ATECC608A are not smart-cards. I would suggest reading though the [datasheet](http://ww1.microchip.com/downloads/en/DeviceDoc/20005927A.pdf) as well as my comments as I try to explain the mapping between PKCS11 concept and ECC hardware.

The SO-PIN does not have an equivalent in the ECC. The ECC only understands keys and can map them in as complicated of a scheme as a user requires. On the other side PKCS11 has no understanding of deriving diversified keys for use in authorizing keys or unwrapping t stored data either. The implementation does have some interfacing for the C_Login api to be able to load the key that would normally be used for ECDH pre shared secret encryption or authorization key. It must be 64 hex characters to match the 32 byte keys used internally.

The remaining questions are all about how you configure a device for your use case. Once you have your use case defined you can configure it to support those different operations. An FAE from Microchip can help you with that.

For the final request however - the ECC devices do not have a master delete feature to completely clear a device. You will have to manually perform operations that will overwrite keys (as in performing a key generation where the device internally will generate a new key in the slot where the old key was).

mickaelchazaux, 2019-06-04:
Hi,

Thank you for the clarification. If I understand correctly, the PKCS#11 support has
to be customized by the integrator.

This explains the "#if PKCS11_TOKEN_INIT_SUPPORT" in pkcs11_token_init and the 
pkcs11-tool --module /lib/libcryptoauth.so --init-token --label 0123EE --so-pin 00000000 --login
sc_dlopen failed: /lib/libcryptoauth.so: undefined symbol: atecc608_config

In fact this initialization process must be customised by the integrator.

My use case is some kind of "micro-HSM" embedded in a router. 
This HSM must be personalized by the end-user of the device. I have no preferred method for this (i2c, crpytoauthlib, or pkcs#11 are okay)
Several software components on the router need to access hardware keys. The standard for this is
PKCS#11. 

So, 
 - on the production line, using cryptoauthlib I write the Configuration Zone according to my needs ( atcab_write_bytes_zone & atcab_lock_config_zone).

 - Then the end-user will initialize the PIN & SO-PIN, using cryptoauthlib (or equivalent in ECC)

 - Next step is key generation, CSR generation & Cert import, again by the end-user (PKCS#11 or crpytoauhtlib).

 - When the certificate expires, this previous process is repeated (automatically with a certificate management protocol or manually).

 - if the SO wants to delete a key, it can be overwritten using cryptoauthlib.

 - Signature production is done by the software components using PKCS#11 only.

Is this feasible with ATECC & cryptoauthlib?

Thank you,

Mickael

closed 2022-11-30 by github-actions[bot]