MicrochipTech/cryptoauthlib #88: ATECC608A + AWS Greengrass

bneff, 2019-06-04:
I'm using an ATECC608A connected over I2C to a Raspberry Pi.  I followed the directions - https://github.com/MicrochipTech/cryptoauthlib/wiki/Greengrass-HSI - and I'm in the final steps of modifying the greengrass configs.  I noticed a couple of issues that I'm "stuck" on.

1. The python library expects the libcryptoauth.so to be in the same directory if you don't pass a lib to load_cryptoauthlib().  I installed this from source and lost some time on this so might be helpful to others.

2. I used the cryptoauthtools to config the 608a device, and I don't see slotUserPin set anywhere in that script, however, the GG config specifies a slotUserPin.  What do I use?

Thanks,

Bryan

bryan-hunt, 2019-06-04:
1. The expectation is that the python library will be installed through pip rather than through sources (based on my understanding the python focusing on pip as being the only recommended installation method) given conflicts between setuptools and pip. As there aren't any instructions for installing the python libraries from sources I was not expecting this to be a common practice. 

I have intended to go back and look into having the python module search through the lib paths to see if it can find a compatible library already installed rather than building and using it's own - still the focus has been on pip. 

Installing the library using pip (with python 3) was quite nice a few months ago but piwheels.org has turned off their build cluster and updated versions of python modules are not getting built. There is no estimate for when the build cluster will go back online so a great many python modules are seeing much more difficult installs on the pi than they were previously. 

2. I'll make an update to the wiki page you cite to be more explicit. The slotUserPin is a dummy value as it isn't being utilized. Since greengrass however requires the field and always provides it to the library it must be a 64 hex character string. You can randomize it or provide the string the way I did in the example configuration in wiki page:

```
slotUserPin = "00112233445566778899AABBCCDDEEFF00112233445566778899AABBCCDDEEFF"
```

bneff, 2019-06-05:
Thank you for the info about the slotUserPin.  I ended up getting the 608A + Greengrass to work, but only with p11-kit-proxy.  If I try to use it without p11-kit-proxy openssl has 2 behaviors I've noticed when trying to generate a CSR.  1. As root it will just hang forever.  2. As non root user it will complain that it failed loading private key.  I'm not against using p11-kit-proxy, but I'd also like to figure out why it doesn't work with cryptoauthlib directly so others don't have to experience this pain.  Any pointers to help solve this?

bryan-hunt, 2019-06-05:
Is this through greengrass or the OpenSSL commandline?

In the greengrass config if the library is used directly the token name is not included in the pkcs11 key string. When using p11-kit the token name has to be included in the pkcs11 key string.

If it’s the OpenSSL command line something must of changed (well plenty always changes in OpenSSL) that is now causing problems. Can you provide for diagnostics:

* OpenSSL version you are using
* openssl.conf
* libp11 version (or commit hash if you built from source)

bryan-hunt, 2019-06-05:
@bneff I have confirmed the lock up issue when directly using OpenSSL due to a missing compile option. I have tested with it restored and all configuration options presented in the wiki article function as expected again.

bneff, 2019-06-05:
@bryan-hunt I just pulled your changes, recompiled, and adjusted the Greengrass config accordingly and can confirm it is working.  Thank you for your help!

commit 5e82b56e (2019-06-05): Resolve #88 mutex issue. The option ATCA_USE_SHARED_MUTEX was dropped when the latest version of cryptoauthlib refactored the platform detection logic in the CMakeLists.txt file
closed 2019-06-24 by bryan-hunt