MicrochipTech/cryptoauthlib #91: Signing device certificate with AT88CKECCSIGNER

copercini, 2019-06-24:
Everything is working with software certificates (thanks @bryan-hunt !), and now we will go to production, but to avoid keys leaks an idea is to use AT88CKECCSIGNER + AT88CKECCROOT, basically replacing https://github.com/MicrochipTech/mbedtls-examples/blob/master/scripts/create_device.py#L134 by the AT88CKECCSIGNER signature, the question is: how?

Is there some API or documentation of how use it to sign certificates/csr ?

bryan-hunt, 2019-06-24:
The Root & Signer kits have specific software and formats that are used. You can find all of the documentation for those kits here: https://www.microchip.com/swlibraryweb/product.aspx?product=cryptoauthprovision

You'll be prompted for your myMicrochip credentials to agree to the license. 

However we do have part provisioning services ourselves where you can delegate signing authority to our HSMs which can simply manufacturing and supply management. Have you contacted your Microchip local sales and support representatives yet? I highly recommend contacting them to discuss supply options as you move closer to production.

copercini, 2019-06-25:
Thanks for the reply!

Do you know if using AT88CKECC-AWS-XSTK or AT88CK590-ND or other signer will be easier to do it (a simple certificate sign, in the same computer of USB signer)? 

AT88CKECCSIGNER seems to have a long way for it (run server program, create a socket to communicate with it, encapsulate the messages in bson, parse the reply and save)

bryan-hunt, 2019-06-25:
Generally this is within the scope of work you would have with a PKI consultancy. The tools provided tend to meet the requirements for small scale PKI systems as they tend to be set up to meet security goals. 

If you don't have a PKI system I would highly suggest pursuing one of the provisioning options from Microchip. Your local sales and support can assist you in setting up those requests.

closed 2019-08-05 by bryan-hunt