MicrochipTech/cryptoauthlib #92: ATECC608A - Assertion Error

neelmitra, 2019-06-24:
HI, 

Can you please help  ? I had it running on my pi and was working fine 2 weeks back, but getting assertion error now. I am planning to run this demo at a conference on wednesday , so will appreciate your quick support. 

i2cdetect works fine and able to recognise the secure element , but the cryptoauth scripts does not work. I reflashed my pi and completed the entire setup again,  but still getting the below errors. 

**The error is as below -** 

pi@raspberrypi:~ $ python cryptoauthtools/python/examples/info.py -i i2c
Traceback (most recent call last):
  File "cryptoauthtools/python/examples/info.py", line 96, in <module>
    info(args.iface, args.device, **parse_interface_params(args.params))
  File "cryptoauthtools/python/examples/info.py", line 48, in info
    assert atcab_init(cfg) == ATCA_SUCCESS
AssertionError

pi@raspberrypi:~ $ python cryptoauthtools/python/examples/config.py -i i2c

Configuring the device with an example configuration
Traceback (most recent call last):
  File "cryptoauthtools/python/examples/config.py", line 241, in <module>
    configure_device(args.iface, args.device, args.i2c, args.gen, **parse_interface_params(args.params))
  File "cryptoauthtools/python/examples/config.py", line 87, in configure_device
    assert atcab_init(cfg) == ATCA_SUCCESS
AssertionError

Thanks for your help in advance.

bryan-hunt, 2019-06-24:
What is the output from i2cdetect?

bryan-hunt, 2019-06-24:
For example on my Raspberry Pi:

```
pi@raspberrypi:~/cryptoauthtools/python/examples $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- 58 -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

Since cryptoauthlib (and the device) use the upper 7 bits to represent the i2c address the above becomes 0xB0 when shifted left one bit. So when I run the example scripts I provide the updated address to them thus:

```
pi@raspberrypi:~/cryptoauthtools/python/examples $ python info.py -i i2c -p slave_address=0xB0

Device Part:
    ATECC608A

Serial number:
    01 23 1E 24 DC 52 1B CD EE

Configuration Zone:
    01 23 1E 24 00 00 60 01 DC 52 1B CD EE C1 35 00
    B0 00 55 01 8F 20 C4 44 87 20 87 20 8F 0F C4 36
    9F 0F 82 20 0F 0F C4 44 0F 0F 0F 0F 0F 0F 0F 0F
    0F 0F 0F 0F 00 00 00 01 00 00 00 00 03 FF FF FF
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    00 00 00 00 00 00 00 00 FF DF 06 40 00 00 00 00
    33 00 1C 00 13 00 13 00 7C 00 1C 00 3C 00 33 00
    3C 00 3C 00 3C 00 30 00 3C 00 3C 00 3C 00 30 00

Check Device Locks
    Config Zone is locked
    Data Zone is locked

Loading Public key

-----BEGIN PUBLIC KEY-----
MFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAERx5/3WCZipmNCmUij/YDM9L9wDFh
trjJobWRExG2LwGFrdDk2yN7x2Zt4jdZg9UlsiO9vd9vCa1p9lugx0VMnQ==
-----END PUBLIC KEY-----

Done
```

neelmitra, 2019-06-24:
Thanks. Below is the output from i2cdetect - 

pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- 35 -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

Now if i run the command as you did , it works - 
pi@raspberrypi:~/cryptoauthtools/python/examples $ python info.py -i i2c -p slave_address=0x6a

Device Part:
    ATECC608A

Serial number:
    01 23 BD D0 F3 27 F7 95 EE

Configuration Zone:
    01 23 BD D0 00 00 60 01 F3 27 F7 95 EE C1 45 00
    6A 00 00 01 85 00 82 00 85 20 85 20 85 20 C6 46
    8F 0F 9F 8F 0F 0F 8F 0F 0F 0F 0F 0F 0F 0F 0F 0F
    0D 1F 0F 0F FF FF FF FF 00 00 00 00 FF FF FF FF
    00 00 00 00 00 00 03 F7 00 69 76 00 00 00 00 00
    00 00 00 00 00 00 00 00 FF FF 0E 60 00 00 00 00
    53 00 53 00 73 00 73 00 73 00 38 00 7C 00 1C 00
    3C 00 1A 00 3C 00 30 00 3C 00 30 00 12 00 30 00

Check Device Locks
    Config Zone is locked
    Data Zone is locked

Loading Public key

-----BEGIN PUBLIC KEY-----
MFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAEWw3CW8aiU3f2/yH//POM8NDQkhh/
ZDaZS7XnqNFSbdwUPfsaxxmOP8nMMRJ+d2N46BJYgsJ811zSYexVMILf7Q==
-----END PUBLIC KEY-----

Done

But if i run without the smave parameter , still getting assertion error.

neelmitra, 2019-06-24:
Below is the 0.conf file i have.

pi@raspberrypi:~/cryptoauthtools/python/examples $ sudo cat /var/lib/cryptoauthlib/0.conf
# Reserved Configuration for a device
# The objects in this file will be created and marked as undeletable
# These are processed in order. Configuration parameters must be comma
# delimited and may not contain spaces

interface = i2c,0x6A
freeslots = 1,2,3

# Slot 0 is the primary private key
object = private,device,0

# Slot 10 is the certificate data for the device's public key
#object = certificate,device,10

# Slot 12 is the intermedate/signer certificate data
#object = certificate,signer,12

# Slot 15 is a public key
object = public,root,15
pi@raspberrypi:~/cryptoauthtools/python/examples $

bryan-hunt, 2019-06-24:
Yes. You have to inform the library that the i2c address is different. It does not scan the i2c bus for things that look like the right device, it instead relies on the user to set the proper field in the configuration structure being passed to atcab_init.

For example in https://github.com/MicrochipTech/cryptoauthtools/blob/6999f32463e520b4d09e7d9f7f7a20149bc3e00c/python/examples/info.py#L48 cfg.cfg.atcai2c.slave_address must be set to 0x6A in your case

The configuration file posted (/var/lib/cryptoauthlib/0.conf) is used by the pkcs11 interface only - it does not affect any other uses.

neelmitra, 2019-06-24:
So are you saying, i cannot hardcode the i2c address in any file and have to pass it for every invocation ? If thats the case , you need to update the github instructions accordingly. Also when i configured 3 weeks back it worked fine, it broke today (atleast i noticed today), so there had been some changes. 

Thanks for the quick turnaround though, i will get going but let me know what you think.

bryan-hunt, 2019-06-24:
You might be interested in the API documentation for the library. For example [atcab_init](https://microchiptech.github.io/cryptoauthlib/html/a00991.html#gacdc4fa3d2f2f6d197af8c10ab1f288b8). 

PKCS11 is a specialized use case and requires external configuration as you see in order to make it work with the existing pkcs11 tooling.

neelmitra, 2019-06-24:
ok will check. A different question, not related to this issue, once some slots are taken , how do i reset the chip to make those slots available again for key generation? 

For example, i have 2 slots taken as below - 

Generating New Keys
    Skipping key pair generation in slot 0: GenKey is disabled
    Skipping key pair generation in slot 1: GenKey is disabled
    Generating key pair in slot 2

What i do this way if 5 of my slots get filledup , how do i rotate these certs or generate new one?

bryan-hunt, 2019-06-24:
Slot usage is defined by configuration. If your configuration allows for modification you can change those slots per what you’ve defined.

Slots hold keys or data and how you use, access, or write them is up to you and is really outside the scope of library support.

You can find quite a few helpful app notes for use cases and examples through https://www.microchip.com/design-centers/security-ics/cryptoauthentication

closed 2019-08-05 by bryan-hunt