MicrochipTech/cryptoauthlib #94: Segmentation fault when using HID

tzewantp, 2019-07-19:
Hi,
I am using the SAMD21 board, and have programmed the firmware on it to allow the USB host (Windows as well as Linux) to enumerate the board as a USB HID device.

On a Windows host, after installing cryptoauthlib, I am able to successfully execute:
`$ python info.py`
I am also able to provision the ATECC608a.

On a Linux x64 host (Ubuntu 18.04) which I wanted to port my development to, I git cryptoauthlib and modified the CMakeLists.txt file as follows:
```
cmake_minimum_required(VERSION 2.6.4)
project(cryptoauth)

# Various Options for Build
option(ATCA_HAL_KIT_HID "Include the HID HAL Driver" ON)
option(ATCA_HAL_I2C "Include the I2C Hal Driver - Linux & MCU only")
option(ATCA_HAL_CUSTOM "Include support for Custom/Plug-in Hal Driver")
option(ATCA_PRINTF "Enable Debug print statements in library" ON)
option(ATCA_PKCS11 "Build PKCS11 Library")
option(ATCA_MBEDTLS "Integrate with mbedtls")
option(ATCA_BUILD_SHARED_LIBS "Build CryptoAuthLib as shared library" ON)
set(BUILD_SHARED_LIBS ${ATCA_BUILD_SHARED_LIBS})
```
Upon completion of the build, I copied the file libcryptoauthlib.so to the python site-packages folder: i.e.,
..../.local/lib/python3.6/site-packages/cryptoauthlib

Yet, when I run info.py (under cryptoauthtools), I get a error 225, followed by a segmentation fault. When I traced the problem, it happened at the following call:
```

ATCA_STATUS atcab_init(ATCAIfaceCfg *cfg)
{
    ATCA_STATUS status = ATCA_GEN_FAIL;

    // If a device has already been initialized, release it
    if (_gDevice)
    {
        atcab_release();
    }

#ifdef ATCA_NO_HEAP
    g_atcab_device.mCommands = &g_atcab_command;
    g_atcab_device.mIface = &g_atcab_iface;
    status = initATCADevice(cfg, &g_atcab_device);
    if (status != ATCA_SUCCESS)
    {
        return status;
    }
    _gDevice = &g_atcab_device;
#else
    _gDevice = newATCADevice(cfg);
    if (_gDevice == NULL)
    {
        return ATCA_GEN_FAIL;  <------------------------ Happens here
    }
#endif
```
What could I have not done correctly during the compiling (i.e. during CMake)? Haven't I already enabled the compiler argument, ATCA_HAL_KIT_HID in the cmake option() command above?

Thanks.

bryan-hunt, 2019-07-19:
Try to use pip to build and install the library on the ubuntu platform first and report the errors from that. The above process is not the proper way to build and install.

tzewantp, 2019-07-20:
Thanks, Bryan for your advice. In my previous post, I was on a Ubuntu 18.04 running as a VM in VirtualBox.

Thinking that the VM may pose other USB issues, I am now trying on a native Ubuntu 16.04 LTS (32 bits) with Python 3.5. It is a relatively old notebook, and is not able to install 64 bits Ubuntu.

In this Ubuntu 16.04 (32 bits), I ran the following command:
`$ pip3 install cryptoauthlib`

It is giving me errors similar to the ones seen in this [thread](https://github.com/MicrochipTech/cryptoauthlib/issues/36).
`Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-mfnlez0h/cryptoauthlib/`

After searching and trying things for hours like upgrading the pip version, it did not help (or could not pinpoint what could help). I am not sure if upgrading to Python 3.6 or using Ubuntu 18.04 LTS would help.

Thanks again and greetings, Tze Wan.

BenUdall-Microchip, 2019-07-22:
Hi @tzewantp, I was able to replicate the problem and it appears to be an issue with an older version of setuptools. Try upgrading it to the latest version:
`pip3 install setuptools --upgrade`

tzewantp, 2019-07-26:
Hi Ben,
Thanks very much for the tip. I will give it a try again.
Regards.

bryan-hunt, 2021-06-21:
This issue has not been reported in many versions of the library so it appears sufficiently out of date.

closed 2021-06-21 by bryan-hunt