MicrochipTech/cryptoauthlib #95: Init with p11tool fails

Xarga19, 2019-07-25:
Running  
`p11tool --initialize`
on a new ATECC608a fails with:  **Error in pkcs11_init:1439: PKCS #11 error.**
I provided the sample config. Running list-all shows a additional public key afterwards.
The full commands are those:
```
test@test:~ $ p11tool --list-all --provider /usr/lib/libcryptoauth.so
Object 0:
        URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=9361D65F12F4;token=0123EE;object=device;type=private
        Type: Private key (EC/ECDSA)
        Label: device
        Flags: CKA_PRIVATE; CKA_SENSITIVE;
        ID:
		
test@test:~ $ p11tool --initialize "pkcs11:serial=9361D65F12F4" --label Testing --provider /usr/lib/libcryptoauth.so
Enter Security Officer's PIN:
Initializing token...
Error in pkcs11_init:1439: PKCS #11 error.

test@test:~ $ p11tool --list-all --provider /usr/lib/libcryptoauth.so
Object 0:
        URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=9361D65F12F4;token=0123EE;object=device;type=private
        Type: Private key (EC/ECDSA-SECP256R1)
        Label: device
        Flags: CKA_PRIVATE; CKA_SENSITIVE;
        ID:

Object 1:
        URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=9361D65F12F4;token=0123EE;object=device;type=public
        Type: Public key (EC/ECDSA-SECP256R1)
        Label: device
        ID:

test@test:~ $ p11tool --export-pubkey --provider /usr/lib/libcryptoauth.so "pkcs11:token=0123EE;object=device;type=private"
warning: --login was not specified and it may be required for this operation.
warning: no --outfile was specified and the public key will be printed on screen.
-----BEGIN PUBLIC KEY-----
MFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAEYKnN1v3MqaDcw88O9peBiMK/QViL
pY9nmaKOrybKE0S7KDpI0Aay/TgTetzJFFxsL69/EUwiFqWhHvk2ab2n+A==
-----END PUBLIC KEY-----
```
Why isn't this working?

bryan-hunt, 2019-07-25:
I am not sure what the error is coming from. The device however is properly initialized. All private keys must have a matching public key object per the PKCS#11 specification. The existence of the matched public key shows that the initialization succeeded. P11tool is accessing the public key to print out the public key of the stored private key in the last step.

agilesai1294, 2019-09-04:
@bryan-hunt  I also not able to Initialize the token for ATECC608A on my raspberry.

raerne, 2019-10-18:
In the `pkcs11` branch I encountered a similar error
```
Error in pkcs11_init:888: PKCS #11 error.
```
The issue was that after initializing the token in `pkcs11_token_init` the I2C slave address is reset to `ATCA_I2C_ECC_ADDRESS` in
https://github.com/MicrochipTech/cryptoauthlib/blob/a0007d2f6c42fddab5dca1575e0f404788829ddc/lib/pkcs11/pkcs11_token.c#L245-L248
 which is defined as
https://github.com/MicrochipTech/cryptoauthlib/blob/a0007d2f6c42fddab5dca1575e0f404788829ddc/lib/atca_config.h#L35

later in `pkcs11_slot_init` this should be caught with 
https://github.com/MicrochipTech/cryptoauthlib/blob/a0007d2f6c42fddab5dca1575e0f404788829ddc/lib/pkcs11/pkcs11_slot.c#L200-L218
but somehow for be (when I check it with gdb) this does not happen.

If I change 
```
ATCA_I2C_ECC_ADDRESS    0xC0
```
the initialization works fine.

twendtland, 2020-04-09:
Any update on this? It's been a while and the problem seems to persist. I am using the p11-tool to initialize, resulting in the same error. However, the public key in slot 0 changes, to I assume _something_ is happening in the ATECx08 (as Bryan has pointed out in a previous comment).

shearl, 2020-06-13:
> If I change
> 
> ```
> ATCA_I2C_ECC_ADDRESS    0xC0
> ```
> 
> the initialization works fine.

@raerne I changed `ATCA_I2C_ECC_ADDRESS` to `0xC0` in `lib/atca_config.h`, but I still get the error. I am using a Raspberry Pi 3 B+. See my setup [here](https://github.com/MicrochipTech/cryptoauthlib/issues/161#issuecomment-643510869) in #161. Any thoughts?

vishalSpintly, 2022-01-29:
Bryan,

I am trying the same with a Rpi4 board with the ATECC608A connected over I2C lines as well.

I find myself in the same situation as you have reported here. I have tried figuring out the problem but I do not find any resources to resolve the issue.

Tried every possible suggestion listed here and elsewhere. searched extensively.
see:

$ p11tool --list-all
warning: no token URL was provided for this operation; the available tokens are:

pkcs11:model=p11-kit-trust;manufacturer=PKCS%2311%20Kit;serial=1;token=System%20Trust
pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=E42440B94379;token=0123EE

$ p11tool --provider /usr/lib/libcryptoauth.so --list-all
Object 0:
URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=E42440B94379;token=0123EE;object=device;type=private
Token '0123EE' with URL 'pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=E42440B94379;token=0123EE' requires user PIN
Enter PIN:
Type: Private key
Label: device
Flags: CKA_PRIVATE; CKA_SENSITIVE;
ID:

Object 1:
URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=E42440B94379;token=0123EE;object=device;type=public
Type: Public key
Label: device
ID:

$ p11tool --provider /usr/lib/libcryptoauth.so --initialize "pkcs11:serial=E42440B94379" --label aws-iot
Enter Security Officer's PIN:
Initializing token...
Error in pkcs11_init:1455: PKCS #11 error.

$ p11tool --list-tokens
Token 0:
URL: pkcs11:model=p11-kit-trust;manufacturer=PKCS%2311%20Kit;serial=1;token=System%20Trust
Label: System Trust
Type: Trust module
Flags: uPIN uninitialized
Manufacturer: PKCS#11 Kit
Model: p11-kit-trust
Serial: 1
Module: p11-kit-trust.so

Token 1:
URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=E42440B94379;token=0123EE
Label: 0123EE
Type: Hardware token, Trust module
Flags: RNG, Requires login, Uninitialized, uPIN uninitialized
Manufacturer: Microchip Technology Inc
Model: ATECC608A
Serial: E42440B94379
Module: /usr/lib/libcryptoauth.so

$ p11tool --export-pubkey --provider /usr/lib/libcryptoauth.so "pkcs11:token=0123EE;object=device;type=private"
note: assuming --login for this operation.
warning: no --outfile was specified and the public key will be printed on screen.
Error in pkcs11_export_pubkey:1397: The requested PKCS #11 object is not available

$ openssl req -engine pkcs11 -key "pkcs11:token=0123EE;object=device;type=private" -keyform engine -new -out new_device.csr -subj "/CN=NEW CSR EXAMPLE"
engine "pkcs11" set.
Found uninitialized token
Specified object not found
Found uninitialized token
Specified object not found
The private key was not found.
PKCS11_get_private_key returned NULL
cannot load Private Key from engine
3069526080:error:80067065:pkcs11 engine:ctx_load_privkey:object not found:eng_back.c:766:
3069526080:error:26096080:engine routines:ENGINE_load_private_key:failed loading private key:../crypto/engine/eng_pkey.c:77:
unable to load Private Key

Can you let me know know if you resolved this problem and if yes, how.
Look forward to your response.

regards,
Vishal

SimoneBongini, 2025-02-27:
> openssl req -engine pkcs11 -key "pkc

hello did you resolve?