MicrochipTech/cryptoauthlib #103: Fake private key PEM file

woelfman, 2019-09-02:
I am looking for instructions on how to generate a fake PEM private key using openssl or p11tool.  I am presently using the [ateccx08 engine](https://github.com/MicrochipTech/cryptoauth-openssl-engine), and I'm working on updating to the pkcs11 engine and latest [cryptoauthlib](https://github.com/MicrochipTech/cryptoauthlib/tree/pkcs11).

Following the [wiki](https://github.com/MicrochipTech/cryptoauthlib/wiki/PKCS11-Linux-Setup) I was able to get my device talking to AWS with openssl.

Now I'm trying to get Mosquitto 1.6 to connect to AWS, which is working with tls_engine ateccx08, and now I am trying to update to tls_engine pkcs11.  I need a fake PEM private key for the bridge_keyfile.

The ateccx08 invocation was:
openssl req -keygen_engine ateccx08 -newkey ec:/tmp/ec-params.pem -sha256 -keyout /tmp/device.key -out /tmp/device.crt -subj '/O=Trimble Navigation Limited/CN=ECDSADEVICE' -passin pass:1234 -passout pass:1234

The pkcs11 invocation I'm trying is:
openssl req -engine pkcs11 -newkey ec:/tmp/ec-params.pem -sha256 -key "pkcs11:model=ATECC508A;manufacturer=Microchip%20Technology%20Inc;serial=E88C88392BC8;token=01232A" -keyform engine -out /tmp/device.csr -subj '/O=Trimble Navigation Limited/CN=ECDSADEVICE'
which appears to succeed without errors but does not generate the private key.  Attached is the log.
[openssl-pkcs11-fake-private-key.log](https://github.com/MicrochipTech/cryptoauthlib/files/3566972/openssl-pkcs11-fake-private-key.log)

nginx has the same concept as mosquitto:
https://www.nginx.com/blog/protecting-ssl-private-keys-nginx-hashicorp-vault/
Configure NGINX to use the vendor’s OpenSSL engine rather than the default software engine:

* ssl_engine vendor-hsm-engine
Rather than using the real private key, configure NGINX to use the vendor‑supplied ‘fake’ key. This key contains a handle that identifies the real key on the HSM device:
* ssl_certificate_key ssl/vendor.private.key
The key may also contain the credentials to access the HSM device, or the credentials may be provided using additional vendor‑specific configuration.

bryan-hunt, 2019-09-02:
OpenSSL to PKCS11 goes through [libp11](https://github.com/OpenSC/libp11) which uses the [RFC 7512](https://tools.ietf.org/html/rfc7512) PKCS11 uri sceme by default to specify the object. If there a “fake key” option available for libp11 I am not aware of it. Cryptoauthlib does not directly interface with OpenSSL.

woelfman, 2019-09-02:
The deprecated cryptoauth-openssl-engine has this feature.  It seems like there may not be an immediate path forward with the new cryptoauthlib and mosquitto until either cryptoauthlib supports the fake key file or mosquitto supports PKCS11 uri.

I appreciate your help.

woelfman, 2019-09-03:
To close loop on this, I believe this to be a bug in Mosquitto (I'm using 1.6.4), where the engine parameters are not being passed to OpenSSL.  I have a local fix that I'll see if I can get it fixed upstream.  With my fix in place `bridge_keyfile` in mosquitto.conf can be set to the uri and the broker connects to AWS.

closed 2019-09-03 by woelfman