MicrochipTech/cryptoauthlib #105: P11tool Initialization of Token Fails

agilesai1294, 2019-09-04:
Hi,

I am trying to follow the steps to set up PKCS on my Raspberry Pi Zero. WHen I followed the step to initialize token I got the following error.

pi@raspberrypi:~ $ p11tool --list-all
warning: no token URL was provided for this operation; the available tokens are:

pkcs11:model=p11-kit-trust;manufacturer=PKCS%2311%20Kit;serial=1;token=System%20Trust
pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=946EEF92A0E7;token=0123EE

pi@raspberrypi:~ $ p11tool --initialize "pkcs11:serial=946EEF92A0E7" --label Testing --provider /usr/lib/libcryptoauth.so
Enter Security Officer's PIN:
Initializing token...
Error in pkcs11_init:1439: PKCS #11 error.

pi@raspberrypi:~ $ p11tool --export-pubkey --provider /usr/lib/libcryptoauth.so "pkcs11:token=0123EE;object=device;type=private"
warning: --login was not specified and it may be required for this operation.
warning: no --outfile was specified and the public key will be printed on screen.
-----BEGIN PUBLIC KEY-----
MFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAEWyEMy85xZUCWkvSRTzNXeA8zP9IZ
1eJ/jVBWaag3sxyVd5d2OyvAIs41bxH4pr06Q8aBT/xuVQmYfslEKjk8Yg==
-----END PUBLIC KEY-----

bryan-hunt, 2019-09-04:
Duplicate of #95. Your token was successfully initialized and ready for use. 

The reported error is a false report and the source of it is being investigated.

closed 2019-09-04 by bryan-hunt

agilesai1294, 2019-09-05:
@bryan-hunt Token isnt initialized.

pi@raspberrypi:~ $ p11tool --initialize "pkcs11:serial=946EEF92A0E7" --label greengrass
Enter Security Officer's PIN:
Initializing token...
Error in pkcs11_init:1439: PKCS #11 error.
pi@raspberrypi:~ $ p11tool --list-all
warning: no token URL was provided for this operation; the available tokens are:

pkcs11:model=p11-kit-trust;manufacturer=PKCS%2311%20Kit;serial=1;token=System%20Trust
pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=946EEF92A0E7;token=0123EE

bryan-hunt, 2019-09-05:
--list-all with multiple tokens only displays the tokens. You need to tell it which token to display the objects for. If the token was not initialized the following command would have failed:

```
 p11tool --export-pubkey --provider /usr/lib/libcryptoauth.so "pkcs11:token=0123EE;object=device;type=private"
```