MicrochipTech/cryptoauthlib #108: Issuer Key ID is not being handled by atcacert_write_cert/atcacert_read_cert

DavidZemon, 2019-09-16:
I am working through the code to confirm this, but it seems as though the `X509v3 Authority Key Identifier` in the certificate template (as generated by `cert2certdef.py` is not being overwritten by the new certificate sent into `atcacert_write_cert()` or `atcacert_read_cert()` (maybe one, maybe both, not sure yet).

Is this something you can easily confirm?

DavidZemon, 2019-09-16:
Yes, it looks like this is getting skipped because all of `std_cert_elements` is skipped in `atcacert_get_device_locs()`. Am I at least on the right track?

Btw, here's the cert definition generated by `cert2certdef.py`, with variable names modified to match our project's coding standards

```
const uint8_t DEVICE_CERTIFICATE_TEMPLATE[] = {
   /* stuff */
};

const atcacert_def_t DEVICE_CERTIFICATE_DEFINITION = {
    .type                   = CERTTYPE_X509,
    .template_id            = 2,
    .chain_id               = 0,
    .private_key_slot       = 0,
    .sn_source              = SNSRC_PUB_KEY_HASH,
    .cert_sn_dev_loc        = {
        .zone      = DEVZONE_NONE,
        .slot      = 0,
        .is_genkey = 0,
        .offset    = 0,
        .count     = 0
    },
    .issue_date_format      = DATEFMT_RFC5280_UTC,
    .expire_date_format     = DATEFMT_RFC5280_GEN,
    .tbs_cert_loc           = {
        .offset = 4,
        .count  = 444
    },
    .expire_years           = 0,
    .public_key_dev_loc     = {
        .zone      = DEVZONE_DATA,
        .slot      = 0,
        .is_genkey = 1,
        .offset    = 0,
        .count     = 64
    },
    .comp_cert_dev_loc      = {
        .zone      = DEVZONE_DATA,
        .slot      = 10,
        .is_genkey = 0,
        .offset    = 0,
        .count     = 72
    },
    .std_cert_elements      = {
        { // STDCERT_PUBLIC_KEY
            .offset = 316,
            .count  = 64
        },
        { // STDCERT_SIGNATURE
            .offset = 460,
            .count  = 74
        },
        { // STDCERT_ISSUE_DATE
            .offset = 163,
            .count  = 13
        },
        { // STDCERT_EXPIRE_DATE
            .offset = 0,
            .count  = 0
        },
        { // STDCERT_SIGNER_ID
            .offset = 155,
            .count  = 4
        },
        { // STDCERT_CERT_SN
            .offset = 15,
            .count  = 16
        },
        { // STDCERT_AUTH_KEY_ID
            .offset = 397,
            .count  = 20
        },
        { // STDCERT_SUBJ_KEY_ID
            .offset = 428,
            .count  = 20
        }
    },
    .cert_elements          = NULL,
    .cert_elements_count    = 0,
    .cert_template          = DEVICE_CERTIFICATE_TEMPLATE,
    .cert_template_size     = sizeof(DEVICE_CERTIFICATE_TEMPLATE)
};
```

DavidZemon, 2019-09-16:
Okay, more reading and `atcacert_get_device_locs()` was the wrong place to look for it. But, I think `atcacert_get_comp_cert()` and `atcacert_set_comp_cert()` are where I should be expecting to find the authority key ID saved/read, and it is not.

DavidZemon, 2019-09-16:
If I am understanding the way `atcacert_get_comp_cert()` works, I don't see how the key IDs could possibly be included in the compressed certificate. The "compressed" certificate does not appear to actually be compressed, merely a concatenation of select fields: 64 bytes for the signature, 4 bytes for a date, 2 bytes for the signer ID, a couple more bytes from I don't even know what that is, and then a final null terminator. No space at all for any extra data.

So now this is looking more like a bug in `cert2certdef.py`, having generated a certificate definition that can't be reused if you ever change the signing key (I invoked it with `cert2certdef.py --device-cert`). And that is even more aggravating because I've looked through that script already and confirmed that it will fail if you run it against any certificate that doesn't have an ID of `FFFF`.

So, if I'm on the right track, where do I report bugs and get help with `cert2certdef.py`?

bryan-hunt, 2019-09-17:
The authority key id is generated by using the public key (of the authority) you pass into the function atcacert_read_cert. If the value is null it assumes the authority key id is embedded in the template.

Please see the API documentation for atcacert_read_cert: https://microchiptech.github.io/cryptoauthlib/a06480.html#gaaf13dfd0ea4e563194cec6e62892b5c4

Certificate definitions can be linked together so you have simple functions that will retrieve the stored public key from the authority certificate and use that for the authority key id. You can see an example of this approach in the pkcs11 interface:

https://github.com/MicrochipTech/cryptoauthlib/blob/pkcs11/lib/pkcs11/pkcs11_cert.c#L46-L113

An example of a chain:
https://github.com/MicrochipTech/cryptoauthlib/blob/pkcs11/app/pkcs11/example_cert_chain.c

DavidZemon, 2019-09-17:
Okay, that got me working. I only recently (in the last week or two) gathered enough information to understand what the authority key was but I wrote the line of code that invokes `atcacert_read_cert()` months ago and completely forgot about that parameter. I have it working now.

closed 2020-06-15 by bryan-hunt