MicrochipTech/cryptoauthlib #111: Compile error on G++: an anonymous union may only have public non-static data members

xaoxtheory, 2019-10-10:
Compiling under G++ 8.3 and I get these errors:

/cryptoauthlib/lib/atca_iface.h:74:16: warning: ‘struct<unnamed struct>::<unnamed union>::ATCAI2C’ invalid; an anonymous union may only have public non-static data members [-fpermissive]
         struct ATCAI2C
                ^~~~~~~
/cryptoauthlib/lib/atca_iface.h:74:16: note: this flexibility is deprecated and will be removed
/cryptoauthlib/lib/atca_iface.h:81:16: warning: ‘struct<unnamed struct>::<unnamed union>::ATCASWI’ invalid; an anonymous union may only have public non-static data members [-fpermissive]
         struct ATCASWI
                ^~~~~~~
/cryptoauthlib/lib/atca_iface.h:86:16: warning: ‘struct<unnamed struct>::<unnamed union>::ATCAUART’ invalid; an anonymous union may only have public non-static data members [-fpermissive]
         struct ATCAUART
                ^~~~~~~~
/cryptoauthlib/lib/atca_iface.h:95:16: warning: ‘struct<unnamed struct>::<unnamed union>::ATCAHID’ invalid; an anonymous union may only have public non-static data members [-fpermissive]
         struct ATCAHID
                ^~~~~~~
/cryptoauthlib/lib/atca_iface.h:104:16: warning: ‘struct<unnamed struct>::<unnamed union>::ATCACUSTOM’ invalid; an anonymous union may only have public non-static data members [-fpermissive]
         struct ATCACUSTOM
                ^~~~~~~~~~
Issue was brought up in issue #43, but was not resolved. Removing the struct names resolves the issue.

balaji-reddy, 2019-10-14:
Here is the patch for C++ compatibility, 
[cryptoauthlib_c_plus_plus_compatible.patch.zip](https://github.com/MicrochipTech/cryptoauthlib/files/3725610/cryptoauthlib_c_plus_plus_compatible.patch.zip)

bryan-hunt, 2020-02-05:
Should be resolved in v3.1.0 (20200205)

closed 2020-06-15 by bryan-hunt