MicrochipTech/cryptoauthlib #115: Openssl CSR Signing fails when a new key is gerenated

raerne, 2019-10-23:
I have the following problem: I would like to Create a new Key in a Slot, then generate a CSR via Openssl/libp11 engine wrapper. This works fine, but when I try to redo this same operation (generating, creating a csr) it fails with an incorrect signature. I'm assuming that the SSL context is somehow not updated about the keychange.

```
    // setup and init engine, ...
    EVP_PKEY *pkey = ENGINE_load_private_key(e, token.c_str(), nullptr, nullptr);
    ENGINE_finish(e);

    if (!X509_REQ_sign(req, pkey, EVP_sha256())) {
        X509_REQ_free(req);
        return -1;
    }

    // ... save request, validation works OK

    if(atcab_genkey(0, nullptr) != ATCA_SUCCESS) {
        X509_REQ_free(req);
        return -1;
    }

    if (!X509_REQ_sign(req, pkey, EVP_sha256())) {
        X509_REQ_free(req);
        return -1;
    }

    // ... save request, validation fails if genkey was called
```

[The full example gist of what I'm trying to do](https://gist.github.com/raerne/1ffbd14fd51e3db2976cba6f904b659f)

What happens that this is not working ?

Thanks for the help.
Ramon

bryan-hunt, 2019-10-23:
Yes this is an OpenSSL and libp11 question. The problem is likely related to the fact you used cryptoauthlib keygen rather than the OpenSSL keygen (which means deleting the key through libp11 and recreating the pair using the engine keygen) As I would have to do much investigation into the workings of OpenSSL to debug it, if you need a prompt answer on it I would suggest the libp11 project.

raerne, 2019-11-13:
I see, since I do not know how (or if it is even possible) to create a new key via Openssl keygen or the pkcs11 interface I solved the problem by reloading the engine via  calling `ENGINE_cleanup()` and then `ENGINE_load_builtin_engines()`.

Thanks for the help.

closed 2022-11-23 by github-actions[bot]