MicrochipTech/cryptoauthlib #116: IAR support

TaniaMirzin, 2019-10-31:
Hi, I see that  atca_compiler.h doesn't supports IAR compiler.
IAR uses __REV,__REV16, __REVSH and __RBIT instead swap functions.
Do you plan to support IAR? 
Except this lack of IAR defines, do you have another  reasons/issues not to support it?

jenia81, 2019-10-31:
IAR support is needed for Pelion Device Management Client project (ARM)

bryan-hunt, 2019-10-31:
It looks like cited intrinsics are ARM specific rather than general purpose IAR ones. We can include these but will have to check the IAR target is ARM. I am slightly concerned with memory alignment however as I can't off the top of my head say that the existing conversions operate on 32 bit word aligned data.

bryan-hunt, 2020-01-24:
@TaniaMirzin @jenia81 Can you confirm if the following addition to compiler.h works correctly in your environment:

```
#elif defined __ICCARM__
/* IAR ARM ------------------------------------------- */
#if __LITTLE_ENDIAN__ == 0
#define ATCA_UINT16_HOST_TO_LE(x)  __REV16(x)
#define ATCA_UINT16_LE_TO_HOST(x)  __REV16(x)
#define ATCA_UINT32_HOST_TO_LE(x)  __REV(x)
#define ATCA_UINT32_HOST_TO_BE(x)  (x)
#define ATCA_UINT32_BE_TO_HOST(x)  (x)
#define ATCA_UINT64_HOST_TO_BE(x)  (x)
#define ATCA_UINT64_BE_TO_HOST(x)  (x)
#else
#define ATCA_UINT16_HOST_TO_LE(x)  (x)
#define ATCA_UINT16_LE_TO_HOST(x)  (x)
#define ATCA_UINT32_HOST_TO_LE(x)  (x)
#define ATCA_UINT32_HOST_TO_BE(x)  __REV(x)
#define ATCA_UINT32_BE_TO_HOST(x)  __REV(x)
#define ATCA_UINT64_HOST_TO_BE(x)  ((uint64_t)__REV((uint32_t)x)<<32 + (uint64_t)__REV((uin32_t)(x>>32)))
#define ATCA_UINT64_BE_TO_HOST(x)  ((uint64_t)__REV((uint32_t)x)<<32 + (uint64_t)__REV((uin32_t)(x>>32)))
#endif
```

TaniaMirzin, 2020-01-30:
@bryan-hunt we need to include  `intrinsics.h` that defines `__REV `functions, and I thinks that uin32_t should be `uint32_t`. With these fixes we able to compile our project.

TaniaMirzin, 2020-01-30:
@bryan-hunt If its ok, when do you plan to open aPR? or commit it?

bryan-hunt, 2020-01-30:
@TaniaMirzin Thank you for reviewing this. A release is being prepared currently so you should see it next week.

bryan-hunt, 2020-02-05:
Should be resolved in v3.1.0 (20200205)

dgreen-arm, 2020-02-06:
We've run into an issue when compiling with IAR using v3.1.0, the error message is
[Error] api_206a.c@508,0: [Pe167]: argument of type "uint16_t *" is incompatible with parameter of type "uint8_t *"

bryan-hunt, 2020-02-07:
@dgreen-arm This is an application file that isn't part of the main library. I would not expect you to have a use case for it. We'll make sure it gets updated - but definitely you shouldn't be including it in your application if you're not using it.

dgreen-arm, 2020-02-10:
I've set these to not be included and the application is building fine with IAR now.

closed 2020-06-15 by bryan-hunt